libfive/libfive #425: Studio.app crashes on startup when built from sources on macOS

szhorvat, 2021-04-24:
### Description

[Description of the issue

Studio.app crashes on startup when built from sources.

[Steps to reproduce]

Build from sources using default options, then launch `studio/Studio.app`

[What did you expect to happen?  What happened?  All the time?]

The app crashes before a window could appear.

[Are there any error messages printed to the console?]

See below. The crash seems to happen when calling `QPlatformFontDatabase::registerFont`.

### Versions

- Operating system: macOS 10.14.6
- `libfive` commit hash: tried both e7486b9b63c1f67509be02399d09b9411ea6c1c2 and a0b883a6fc5d4e3b4b70b4ed1dccf0a21b2d5c46
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.15.2

### Additional Information

<details><summary>CMake config output</summary>

```
-- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /opt/local/include (found version "1.71.0")
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2")
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.7
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.9
-- Found ZLIB: /opt/local/lib/libz.dylib (found version "1.2.11")
-- Found PNG: /opt/local/lib/libpng.dylib (found version "1.6.37")
-- Found Python3: /opt/local/bin/python3.9 (found version "3.9.4") found components: Interpreter Development Development.Module Development.Embed
Checking dependencies:
  libfive:		✓
  Guile bindings:	✓
  Python bindings:	✓
  Studio:		✓   (Python + Guile)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/szhorvat/Repos/libfive/build
```
</details>

<details><summary>Crashlog saved by macOS</summary>

```
Process:               Studio [9253]
Path:                  /Users/USER/*/Studio.app/Contents/MacOS/Studio
Identifier:            com.impraxical.studio
Version:               0.0.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Studio [9253]
User ID:               503

Date/Time:             2021-04-24 09:57:37.942 +0200
OS Version:            Mac OS X 10.14.6 (18G8022)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        21404924-0EDA-09D8-EBDC-F3AC96C2FE86

Sleep/Wake UUID:       4083CD70-B856-44A9-A767-F0E5B45D9828

Time Awake Since Boot: 350000 seconds
Time Since Wake:       82000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [9253]

VM Regions Near 0:
--> 
    __TEXT                 000000010a8fb000-000000010a953000 [  352K] r-x/r-x SM=COW  /Users/USER/*/Studio.app/Contents/MacOS/Studio

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtGui          	0x000000010b6a25f4 0x10b60c000 + 615924
1   org.qt-project.QtGui          	0x000000010b6a2aae 0x10b60c000 + 617134
2   org.qt-project.QtGui          	0x000000010b70456e QPlatformFontDatabase::registerFont(QString const&, QString const&, QString const&, QFont::Weight, QFont::Style, QFont::Stretch, bool, bool, int, bool, QSupportedWritingSystems const&, void*) + 68
3   libqcocoa.dylib               	0x00000001102f1078 0x11028d000 + 409720
4   libqcocoa.dylib               	0x00000001102f272e 0x11028d000 + 415534
5   org.qt-project.QtGui          	0x000000010b6a2ffb 0x10b60c000 + 618491
6   org.qt-project.QtGui          	0x000000010b6a2f29 0x10b60c000 + 618281
7   org.qt-project.QtGui          	0x000000010b698b9e QFontEngineMulti::ensureFallbackFamiliesQueried() + 70
8   org.qt-project.QtGui          	0x000000010b698c5a QFontEngineMulti::ensureEngineAt(int) + 38
9   org.qt-project.QtGui          	0x000000010b6995ba QFontEngineMulti::getGlyphBearings(unsigned int, double*, double*) + 38
10  org.qt-project.QtGui          	0x000000010b6bce48 0x10b60c000 + 724552
11  org.qt-project.QtGui          	0x000000010b6bb013 QTextLine::layout_helper(int) + 2717
12  org.qt-project.QtWidgets      	0x000000010b2d8484 QPlainTextDocumentLayout::layoutBlock(QTextBlock const&) + 354
13  org.qt-project.QtWidgets      	0x000000010b2d821c QPlainTextDocumentLayout::blockBoundingRect(QTextBlock const&) const + 76
14  org.qt-project.QtWidgets      	0x000000010b2d98c9 0x10b1a2000 + 1276105
15  org.qt-project.QtWidgets      	0x000000010b2f9980 0x10b1a2000 + 1407360
16  org.qt-project.QtWidgets      	0x000000010b2fad48 QWidgetTextControl::cursorRect(QTextCursor const&) const + 74
17  org.qt-project.QtWidgets      	0x000000010b2dad5e 0x10b1a2000 + 1281374
18  org.qt-project.QtWidgets      	0x000000010b2dd392 0x10b1a2000 + 1291154
19  org.qt-project.QtWidgets      	0x000000010b2db94c QPlainTextEdit::cursorRect() const + 44
20  org.qt-project.QtWidgets      	0x000000010b2dac77 0x10b1a2000 + 1281143
21  org.qt-project.QtWidgets      	0x000000010b2de80e 0x10b1a2000 + 1296398
22  org.qt-project.QtWidgets      	0x000000010b2f54ac 0x10b1a2000 + 1389740
23  com.impraxical.studio         	0x000000010a913bb1 Studio::Editor::loadDefaultScript() + 49 (qstring.h:1307)
24  com.impraxical.studio         	0x000000010a91315b Studio::Editor::Editor(Studio::Language::Type) + 2139 (qstring.h:1307)
25  com.impraxical.studio         	0x000000010a923f08 Studio::Window::Window(Studio::Arguments) + 152
26  com.impraxical.studio         	0x000000010a906abc Studio::App::App(int&, char**) + 316 (qscopedpointer.h:155)
27  com.impraxical.studio         	0x000000010a905775 main + 149 (main.cpp:51)
28  libdyld.dylib                 	0x00007fff698e03d5 start + 1
```
</details>

mkeeter, 2021-04-24:
Weird, I can't reproduce this one.  I added a debug message in 44652a9729b2fcf9a0d6b44100eee04c27e7a84f that will print if `QFontDataBase::addApplicationFont` fails, but it _shouldn't_, because the font is bundled in the Qt resource file.  Can you update and see what happens there?

szhorvat, 2021-04-24:
This is the contents of `Studio.app`:

```
$ tree Studio.app/
Studio.app/
└── Contents
    ├── Info.plist
    └── MacOS
        └── Studio

2 directories, 2 files
```

Maybe I am not building correctly. I just did:

```
cmake ..
ninja
```

from the build directory, and that's it. Do I need to do anything else to copy the font file into the app bundle?

If I try to run the executable from the app bundle directly (with the intention to be able to see messages printed to the terminal), I get:

```
$ Studio.app/Contents/MacOS/Studio
guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
QMutex: destroying locked mutex
```

mkeeter, 2021-04-24:
That is the correct thing to see in `Studio.app` after a fresh build, and I see the same:
```
➜  build git:(master) tree studio/Studio.app/
studio/Studio.app/
└── Contents
    ├── Info.plist
    └── MacOS
        └── Studio

2 directories, 2 files
```

(in case you're curious, the font is bundled using [Qt's resource system](https://doc.qt.io/qt-5/resources.html), so it's actually included in the executable)

Have you moved the Studio executable out of the `build` directory?  The `Studio.app` bundle **is not standalone** after `ninja` is run – it expects to be in the `build` directory, and will be unhappy otherwise.  I invoke it from the build directory with
```
➜  build git:(master) ./studio/Studio.app/Contents/MacOS/Studio
```

To make it standalone, run [`deploy.sh`](https://github.com/libfive/libfive/blob/master/studio/deploy/mac/deploy.sh) from the `studio/deploy/mac` directory.  This will copy in all of the `dylibs` and resources required to deploy the `.app` bundle.  This is untested on machines other than mine, since I just use it for the binaries on [libfive.com](https://libfive.com/download).

szhorvat, 2021-04-24:
> ➜  build git:(master) ./studio/Studio.app/Contents/MacOS/Studio

Yes, this works fine.

What does not work is:

 - Opening through the Finder
 - `open studio/Studio.app`
 - `cd studio && Studio.app/Contents/MacOS/Studio`

So it is not sufficient to avoid moving it. One must also (1) launch the executable from inside the app bundle directly (instead of `open`) and (2) have the current directory be `build` when doing so.

szhorvat, 2021-04-24:
I was looking to make Studio and libfive available through MacPorts. In that case, the app bundle would be moved, but it would not contain all Qt libraries. Instead, it would link to Qt located within the MacPorts installation directory. The libfive dylib should also be installed to there. 

However, I now realize that I probably don't have the knowledge to make all this work. Ideally, `libfive` and Studio would be separated into two packages, and Studio would like to an already installed (`ninja install`) libfive. I'll experiment a bit, but I might give up on this.

mkeeter, 2021-04-24:
Opening through Finder should be fixed in d8cf2d9c96c97041d71bc880e401fe00f5f45b97, so I'm going to close this issue.

Deploying through Macports would definitely be a challenge from a packaging perspective, as you could split it into up to four pieces:
- `libfive.dylib` and `libfive-stdlib.dylib`
- Python bindings
- Guile bindings
- The `Studio` application itself

This complexity is why I wrote the `deploy.sh` packaging script, which deploys everything in a single `.app` bundle.  This is easy to distribute, but is obviously less efficient – it includes a whole Python and Guile runtime 😆

closed 2021-04-24 by mkeeter

szhorvat, 2021-04-24:
Is it possible to build the Python or Guile bindings based on an existing libfive installation (rather than always compiling libfive first)?

mkeeter, 2021-04-24:
Perhaps, but it would require a decent amount of restructuring.

There are separate `CMakeLists` for each subdirectory, but the top-level `CMakeLists.txt` does non-trivial work (settings flags, finding packages, etc) before including subdirectories.

szhorvat, 2021-04-25:
@mkeeter What is the license of the Python bindings? libfive is MPLv2, Guile bindings and Studio are GPLv2+, what about the Python bindings?

szhorvat, 2021-04-25:
FYI here's a draft PR for inclusion into MacPorts: https://github.com/macports/macports-ports/pull/10791