libfive/libfive #426: Script files not loading in Studio on MacOS 10.15

leokeba, 2021-04-26:
### Description

Wether it's loading script files through finder, or through file menu > open, nothing happens and I just get the default script. I can, however, directly paste the text from the file, and then it works like expected. Is there anything specific I can do to investigate further ?

### Versions

- Operating system: MacOS 10.15.7
- `libfive` commit hash: latest release, i get this from the "about" menu : a0b883a
  - Is this the latest commit to `master`?  no
- Qt version (if applicable): Don't know

### Additional Information

Is there a log that I can probe somewhere to get more info about this behaviour ?

Thanks, and keep up the good work :)

mkeeter, 2021-04-26:
Looks like this is a dupe of #411 , so I'll rebuild a fresh app and update the website.

mkeeter, 2021-04-26:
Updated!

closed 2021-04-26 by mkeeter

leokeba, 2021-04-26:
Thanks a lot!