libfive/libfive #427: Most Scheme examples don't work

szhorvat, 2021-04-26:
### Description

[Description of the issue]

Most Scheme examples for Studio don't work in the version I compiled from the current master.

[Steps to reproduce]

For example, load the tutorial.

[image]

[What did you expect to happen?  What happened?  All the time?]

[Are there any error messages printed to the console?]

I get the error:

```
In function make-shape:
wrong-type-arg: Opcode sqrt takes 1 shape argument
```

This does not happen in the version downloaded from libfive.com

### Versions

- Operating system: macOS 10.14
- `libfive` commit hash: d8cf2d9c96c97041d71bc880e401fe00f5f45b97
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.15.2

### Additional Information

[Any additional information, configuration or data that might be necessary to reproduce the issue.]

Several other of the included examples don't work either. The function `square` shows a similar error message.

### Common questions

[If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?]

Yes, except for what is noted in #424

szhorvat, 2021-04-26:
If I build a0b883a6 from source, it works fine.

mkeeter, 2021-04-26:
Thanks, fixed – it was a silly parenthesis issue.

commit f471c12e (2021-04-26): Fix #427
closed 2021-04-26 by mkeeter