libfive/libfive #430: Deprecation block makes C compiler fall over

virtualritz, 2021-05-04:
This is from line 236 in libfive.h
```c
[[deprecated("use libfive_tree_nullary instead")]]
libfive_tree libfive_tree_nonary(int op);
```

Which makes clang fall over with:
```
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:236:2: error: expected expression
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int]
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:1: error: redefinition of 'libfive_tree' as different kind of symbol
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:188:31: note: previous definition is here
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:13: error: expected ';' after top level declarator
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:236:2: error: expected expression, err: true
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:1: warning: type specifier missing, defaults to 'int' [-Wimplicit-int], err: false
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:1: error: redefinition of 'libfive_tree' as different kind of symbol, err: true
  /Users/moritz/code/libfive/build/libfive/include/libfive.h:237:13: error: expected ';' after top level declarator, err: true
```

I think this `[[...]]` stuff is C++14 and later but a C compiler  can't understand it?

mkeeter, 2021-05-05:
Does this work with
```
__attribute__((deprecated("deprecated", "libfive_tree_nullary")))
libfive_tree libfive_tree_nonary(int op);
```
?

virtualritz, 2021-05-05:
Yes, that fixes it for me.

closed 2021-05-06 by mkeeter

dbordak, 2021-05-09:
This change broke it for me w/ gcc

```
error: wrong number of arguments specified for ‘deprecated’ attribute
  237 | libfive_tree libfive_tree_nonary(int op);
      |                                        ^
```

(btw, I don't know too much about debugging C compilers, I'm just using the AUR package)

virtualritz, 2021-05-09:
I suggest using my [pragmatic fix](https://github.com/virtualritz/libfive/commit/d6730fbce796b836fedd68e29325f6ab74e0d4b7) in the meantime.

It just moves the whole resp. signature and its deprecation hint into a section guarded by `#ifdef __cplusplus`.

mkeeter, 2021-05-09:
Yeah, I'd need to know what version of GCC you're using vs @virtualritz .

I'm targeting "whatever is in the latest Ubuntu LTS" for all packages, which means `9.3` if I'm reading [this](https://packages.ubuntu.com/search?keywords=gcc) right

TheCatster, 2021-05-09:
I've tried copy/pasting in virtualritz's fix into the cloned repository (from master, 15 minutes ago,) into the AUR packages directory, and simply cloning virtualritz's fork. All result in this same error. My GCC version is 10.20.0

mkeeter, 2021-05-09:
New plan – I deleted the deprecated function + heater in ae8ade5ada97af319f6ea07f06bd8242336171cd.  No one is actively writing bindings against the C API other than @virtualritz (that I know of), and he's in this thread 😆

TheCatster, 2021-05-10:
That worked great! Sad to hear no one's using the C API, but at least it supports the newer GCCs now. Thank you!

commit 9aa20520 (2021-05-06): Switch to C99-compatible syntax for deprecation (fixes #430)