libfive/libfive #434: [C-API] Why is the 'quality' parameter only exposed in libfive_tree_save_meshes()

virtualritz, 2021-05-09:
See subject.

mkeeter, 2021-05-09:
Not sure, probably just a historical artifact.

The C API is basically an accumulation of "whatever I want to call from a higher-level language", so it's not particularly orthogonal.

My one justification is that `libfive_tree_save_meshes` is the "fancy" function which exposes more arguments; it's possible to use it to export one mesh by passing `{tree, NULL}` as the first argument.

virtualritz, 2021-05-10:
I am more interested in obtaining a mesh in memory from the lib. So I see two options:

1. I can add the `quality` knob to all functions in the C API where this makes sense (probably all that also have `resolution`.
2. I can add a `settings` struct to the C API and pass that to all resp. functions instead. The meshing algorithm is e.g. exposed in Studio but not at all in the C API.