libfive/libfive #437: Some way for people to give you money

traverseda, 2021-05-10:
Now that python support is here I'd throw a few dollars a month towards a patreon or equivalent, for a long long time.

sene-je, 2022-06-22:
Since they didn't? take it, how about donating it in his name to Greta Thunberg's charity, ask the Coding Horror guy (can't remember his name offhand, Jeff, uh still can't remember) if he knows of any (I'm not sure he needs any himself), there is the director of a visualization lab, whose last email to me was "Do you have money for my students?"  to which you can reply, "Sene Je (translated from Ido to English) said, 'We do now, in Libfive's name.":

www.uwyo.edu/cosc/cosc-directory/abanic/index.html

Or one of the other charities (in Libfive's name) I have listed here:

https://github.com/github-community/community/discussions/19179

I don't need any, I'm more like Jeff Atwood (remembered) and you, out of which money flows to charity.

sene-je, 2022-06-22:
Thanks for the charity tip!

What then needs money to sustain here?  It's a free repo, externally funded owners apparently.

My idea with Dr. Banic was to pay her students to do work.  Maybe you could have them do some little jobs her...like...cleaning up some of these still open issues for a little Patron [sp?] money.  Pay them to learn everything the owners know so we don't need them anymore. [humor?]

Or again, ask Jeff Atwood the reverse question?  

I hope no hard feelings with *the Libfive owners*, but Jeff and I are like creative supernovas and really shouldn't be on the same email thread together.  But he is extremely nice--it was like Matt Grohl meeting his grunge influence when I talked to Jeff.  Ultimately my email went up the chain to Stroustrup's org, my alma mater, and he responded to my criticism of his X windows crashing toy language with a McD's add campaign--"MMM, mmm...You know you want to buy our machine produced breakfast muffin sandwiches."  Long, long, story.

The owners here are really nice too, but I don't need them any more--I've got my own FGM (functional geometric machine) called Machinate Scheme, a Machiavellian Scheme (M. Python reference to Kissenger song--he's still alive and doing diplomatic work), it's verrry nice you see.  It is running on my 2007 Dell Inspiron, closed source, with sphere tracing, DC, octrees, simple built in image library, no external dependencies except a Posix Unix and C.  Very lightweight and energy efficient.  I did it just by turning off the internet (I mean completely, Lot's wife) and sitting down and doing it, 2 times in one year (throw the first one away principle, er second).  It is similar to Ruckus, without the (again giant) dependency of Racket.  I'm trying to design a new graphics architecture to run it on, maybe even with recycled materials (serious effort, not kidding)

Boost interval library ported from C++ to Scheme in 2 afternoons, no need at all for Eigen (I can't even figure out what "the owners" need it for anyway.)  It compiles in 30 seconds, 30 more with FemtoLisp^H^H^H^H^H^HMachinate.  The AST is in prefix Scheme where it belongs, like that non-Dylan infix Scheme.  It can generate GPU code, S-assembly (if I can ever sit down to write it), native, anything you can code gen to.  Remember your Knuth, "Premature optimization is the root of all evil."  And Dykstra who wrote his code and notes down on paper.

Regards. (I typed this all in one sitting, streaming from my brain...meet Jeff and you'll see what I mean with keeping boron rods between us.)  The owners are going to kick us off the channel if we don't stop, so I hope you find a home for your bucks.

mkeeter, 2022-06-22:
This is going a little off the rails, so I'm going to hide the unrelated comments in this thread and #209.

@sene-je, your writing style seems familiar, so I think we've interacted before (under your previous aliases, which appear to be deleted now).  Please try to keep stuff on topic!