libfive/libfive #438: Libfive-studio segfaults with `move(sphere(1),[])`

traverseda, 2021-05-11:
### Description

While playing around with libfive studio last night I entered this snippet, and studio segfaulted.

```python3
from libfive.stdlib import *
move(sphere(1),[])
```

This probably shouldn't segfault the entire studio application, if possible, as a segfault will kill all unsaved work.

### Versions

- Operating system: Arch Linux kde/plasma
- `libfive` commit hash: ae8ade5
  - Is this the latest commit to `master`?  yes

mkeeter, 2021-05-11:
Interesting – it looks like `ctypes.Structure` will happily construct a `struct` without any arguments, even if it's supposed to take X/Y/Z coordinates (`tvec3` [here](https://github.com/libfive/libfive/blob/master/libfive/bind/python/libfive/ffi.py#L70-L73)).  This obviously blows up, since the fields are set to `None` by default.

mkeeter, 2021-05-18:
This should be fixed now – turns out you can override the default `ctypes.Structure` constructor, so I made one that takes the right number of arguments.

commit cdc363d0 (2021-05-18): Add explicit constructor for tvec2/3 (fixes #438)
closed 2021-05-18 by mkeeter