libfive/libfive #440: Guile bindings build failure on OS/X 10.14.6

jcasu, 2021-05-13:
### Description

Trying to build libfive/studio
Guile bindings (& hence studio) fail to build

[Description of the issue]
[Steps to reproduce]

(base) Galactica:libfive-master john$ mkdir build ; cd build
(base) Galactica:build john$ brew install cmake pkg-config eigen libpng boost guile python3 qt
(base) Galactica:build john$ cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt\@5/5.15.2/ ..
(base) Galactica:build john$ make -j 4

[What did you expect to happen?  What happened?  All the time?]

I expected the build to run to completion.

[Are there any error messages printed to the console?]

[ 54%] Linking CXX shared library libfive-stdlib.dylib
[ 54%] Built target libfive-stdlib
[ 54%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/contours.cpp.o
[ 55%] Generating libfive/kernel.go
Backtrace:
In ice-9/psyntax.scm:
  1230:36 19 (expand-top-sequence (#<syntax:kernel.scm:21:0 (#<syn…>) …)
  1222:19 18 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   259:10 17 (parse _ (("placeholder" placeholder)) (()) _ c&e (# #) #)
In ice-9/boot-9.scm:
  3927:20 16 (process-use-modules _)
   222:29 15 (map1 (((oop goops)) ((rnrs bytevectors)) ((srfi #)) # …))
   222:29 14 (map1 (((rnrs bytevectors)) ((srfi srfi-28)) ((# #)) # …))
   222:29 13 (map1 (((srfi srfi-28)) ((system foreign)) ((# lib)) #))
   222:29 12 (map1 (((system foreign)) ((libfive lib)) ((libfive …))))
   222:17 11 (map1 (((libfive lib)) ((libfive vec))))
  3928:31 10 (_ ((libfive lib)))
  3326:17  9 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8  8 (_ _)
In ice-9/boot-9.scm:
  3252:13  7 (_)
In ice-9/threads.scm:
    390:8  6 (_ _)
In ice-9/boot-9.scm:
  3536:20  5 (_)
   2835:4  4 (save-module-excursion #<procedure 10bb6e0f0 at ice-9/b…>)
  3556:26  3 (_)
In unknown file:
           2 (primitive-load-path "libfive/lib" #<procedure 10a78c3e…>)
In ice-9/boot-9.scm:
   2007:7  1 (error _ . _)
  1685:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1685:16: In procedure raise-exception:
Could not find libfive shared library
make[2]: *** [libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 56%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/deck.cpp.o
[ 56%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/dual.cpp.o

### Versions

- Operating system: [here]

OSX 10.14.6

- `libfive` commit hash: [here]
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): [here]

### Additional Information

Any additional information, configuration or data that might be necessary to reproduce the issue.

### Common questions
[Delete this section if not relevant]

If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?

If this is a graphics issue on Linux, report the results of updating your graphics drivers (and switching to binary graphics drivers if applicable).

If there are error messages in the console, have you searched for them and followed up on any promising leads?

mkeeter, 2021-05-14:
If you run `make libfive/src/libfive.dylib`, then re-run `make`, does it work?  I wonder if the Guile build step doesn't have its dependencies correctly wired up.

jcasu, 2021-05-14:
Alas, it does not work:

(base) Galactica:build john$ make libfive/src/libfive.dylib
(base) Galactica:build john$ make
[ 44%] Built target libfive
[ 46%] Built target libfive-stdlib
[ 69%] Built target libfive-test
[ 70%] Generating libfive/kernel.go
Backtrace:
In ice-9/psyntax.scm:
  1230:36 19 (expand-top-sequence (#<syntax:kernel.scm:21:0 (#<syn…>) …)
  1222:19 18 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   259:10 17 (parse _ (("placeholder" placeholder)) (()) _ c&e (# #) #)
In ice-9/boot-9.scm:
  3927:20 16 (process-use-modules _)
   222:29 15 (map1 (((oop goops)) ((rnrs bytevectors)) ((srfi #)) # …))
   222:29 14 (map1 (((rnrs bytevectors)) ((srfi srfi-28)) ((# #)) # …))
   222:29 13 (map1 (((srfi srfi-28)) ((system foreign)) ((# lib)) #))
   222:29 12 (map1 (((system foreign)) ((libfive lib)) ((libfive …))))
   222:17 11 (map1 (((libfive lib)) ((libfive vec))))
  3928:31 10 (_ ((libfive lib)))
  3326:17  9 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8  8 (_ _)
In ice-9/boot-9.scm:
  3252:13  7 (_)
In ice-9/threads.scm:
    390:8  6 (_ _)
In ice-9/boot-9.scm:
  3536:20  5 (_)
   2835:4  4 (save-module-excursion #<procedure 10d9b9180 at ice-9/b…>)
  3556:26  3 (_)
In unknown file:
           2 (primitive-load-path "libfive/lib" #<procedure 10c52196…>)
In libfive/lib.scm:
     43:2  1 (_)
In ice-9/boot-9.scm:
  1685:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1685:16: In procedure raise-exception:
Could not find libfive shared library
make[2]: *** [libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2
make: *** [all] Error 2

mkeeter, 2021-05-14:
Can you try building with [`ninja`](https://ninja-build.org/) instead?

This means deleting + recreating your build folder, running CMake with `-GNinja`, then calling `ninja` instead of `make` to actually build.

From your current directory, does `libfive/src/libfive.dylib` exist?  The [Scheme code](https://github.com/libfive/libfive/blob/master/libfive/bind/guile/libfive/lib.scm#L26-L43) to find the library is pretty simple, so it's hard to understand what could go wrong.

virtualritz, 2021-05-14:
Just tried with `ninja -v`. Same result:

```                                                                                                                                        138.199.36.153   100% 
[1/16] cd /Users/moritz/code/crates/libfive/libfive-sys/libfive/build && guild compile -L/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile /Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/transforms.scm && /usr/local/Cellar/cmake/3.20.2/bin/cmake -E copy /Users/moritz/.cache/guile/ccache/3.0-LE-8-4.5/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/transforms.scm.go /Users/moritz/code/crates/libfive/libfive-sys/libfive/build/libfive/bind/guile/libfive/stdlib/transforms.go
FAILED: libfive/bind/guile/libfive/stdlib/transforms.go
cd /Users/moritz/code/crates/libfive/libfive-sys/libfive/build && guild compile -L/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile /Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/transforms.scm && /usr/local/Cellar/cmake/3.20.2/bin/cmake -E copy /Users/moritz/.cache/guile/ccache/3.0-LE-8-4.5/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/transforms.scm.go /Users/moritz/code/crates/libfive/libfive-sys/libfive/build/libfive/bind/guile/libfive/stdlib/transforms.go
Backtrace:
In system/base/compile.scm:
   261:27 19 (_ _ _)
In ice-9/boot-9.scm:
   2835:4 18 (save-module-excursion #<procedure 101ca16e0 at languag…>)
In language/scheme/compile-tree-il.scm:
    31:15 17 (_)
In ice-9/psyntax.scm:
  1230:36 16 (expand-top-sequence (#<syntax:transforms.scm:11:0 (#…>) …)
  1222:19 15 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   259:10 14 (parse _ (("placeholder" placeholder)) (()) _ c&e (# #) #)
In ice-9/boot-9.scm:
  3927:20 13 (process-use-modules _)
   222:29 12 (map1 (((system foreign)) ((libfive lib)) ((# #)) ((…))))
   222:17 11 (map1 (((libfive lib)) ((libfive kernel)) ((libfive …))))
  3928:31 10 (_ ((libfive lib)))
  3326:17  9 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8  8 (_ _)
In ice-9/boot-9.scm:
  3252:13  7 (_)
In ice-9/threads.scm:
    390:8  6 (_ _)
In ice-9/boot-9.scm:
  3536:20  5 (_)
   2835:4  4 (save-module-excursion #<procedure 100f72e10 at ice-9/b…>)
  3556:26  3 (_)
In unknown file:
           2 (primitive-load-path "libfive/lib" #<procedure 101ca758…>)
In libfive/lib.scm:
     43:2  1 (_)
In ice-9/boot-9.scm:
  1685:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1685:16: In procedure raise-exception:
Could not find libfive shared library
[2/16] cd /Users/moritz/code/crates/libfive/libfive-sys/libfive/build && guild compile -L/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile /Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/text.scm && /usr/local/Cellar/cmake/3.20.2/bin/cmake -E copy /Users/moritz/.cache/guile/ccache/3.0-LE-8-4.5/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/text.scm.go /Users/moritz/code/crates/libfive/libfive-sys/libfive/build/libfive/bind/guile/libfive/stdlib/text.go
FAILED: libfive/bind/guile/libfive/stdlib/text.go
cd /Users/moritz/code/crates/libfive/libfive-sys/libfive/build && guild compile -L/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile /Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/text.scm && /usr/local/Cellar/cmake/3.20.2/bin/cmake -E copy /Users/moritz/.cache/guile/ccache/3.0-LE-8-4.5/Users/moritz/code/crates/libfive/libfive-sys/libfive/libfive/bind/guile/libfive/stdlib/text.scm.go /Users/moritz/code/crates/libfive/libfive-sys/libfive/build/libfive/bind/guile/libfive/stdlib/text.go
Backtrace:
In system/base/compile.scm:
   261:27 19 (_ _ _)
In ice-9/boot-9.scm:
   2835:4 18 (save-module-excursion #<procedure 1093923e0 at languag…>)
In language/scheme/compile-tree-il.scm:
    31:15 17 (_)
In ice-9/psyntax.scm:
  1230:36 16 (expand-top-sequence (#<syntax:text.scm:11:0 (#<synta…>) …)
  1222:19 15 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   259:10 14 (parse _ (("placeholder" placeholder)) (()) _ c&e (# #) #)
In ice-9/boot-9.scm:
  3927:20 13 (process-use-modules _)
   222:29 12 (map1 (((system foreign)) ((libfive lib)) ((# #)) ((…))))
   222:17 11 (map1 (((libfive lib)) ((libfive kernel)) ((libfive …))))
  3928:31 10 (_ ((libfive lib)))
  3326:17  9 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8  8 (_ _)
In ice-9/boot-9.scm:
  3252:13  7 (_)
In ice-9/threads.scm:
    390:8  6 (_ _)
In ice-9/boot-9.scm:
  3536:20  5 (_)
[... 138 lines cut ...]
   222:29 12 (map1 (((system foreign)) ((libfive lib)) ((libfive …))))
   222:17 11 (map1 (((libfive lib)) ((libfive vec))))
  3928:31 10 (_ ((libfive lib)))
  3326:17  9 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8  8 (_ _)
In ice-9/boot-9.scm:
  3252:13  7 (_)
In ice-9/threads.scm:
    390:8  6 (_ _)
In ice-9/boot-9.scm:
  3536:20  5 (_)
   2835:4  4 (save-module-excursion #<procedure 10b74ecf0 at ice-9/b…>)
  3556:26  3 (_)
In unknown file:
           2 (primitive-load-path "libfive/lib" #<procedure 10b762d6…>)
In libfive/lib.scm:
     43:2  1 (_)
In ice-9/boot-9.scm:
  1685:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1685:16: In procedure raise-exception:
Could not find libfive shared library
ninja: build stopped: subcommand failed.
```

jcasu, 2021-05-14:
just confirming that using ninja also fails.

mkeeter, 2021-05-15:
I've upgraded Guile and can now reproduce, so I'll investigate!

mkeeter, 2021-05-15:
This should be fixed in 68d5b46cdff2e063d2b72a163eeccd027f8e9dea – it appears to be a bug in *Guile*, so I've deployed a workaround and reported it to the mailing list.

closed 2021-05-15 by mkeeter

zeeyang, 2021-05-16:
I was having build issues so I put together a [CI build](https://github.com/libfive/libfive/pull/443). Unfortunately this fix does not work in catalina (CI image). However it does work on my local machine (big sur).

zeeyang, 2021-05-16:
@mkeeter personally python binding works fine for me. I wouldn't mind waiting for guile to fix this upstream.