libfive/libfive #442: Studio build fails on macOS

virtualritz, 2021-05-14:
I am trying to build Studio w/o Guild bindings (#440).

```
cmake ../ -GNinja -DBUILD_STUDIO_APP=ON -DBUILD_GUILE_BINDINGS=OFF -DBUILD_PYTHON_BINDINGS=ON -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt@5/5.15.2
```

```
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /usr/local/lib/cmake/Boost-1.75.0/BoostConfig.cmake (found version "1.75.0")
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2")
-- Checking for module 'guile-2.2>=2.2.1'
--   No package 'guile-2.2' found
-- Checking for module 'guile-3.0>=3.0.0'
--   Found guile-3.0, version 3.0.7
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.9
-- Found ZLIB: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/usr/lib/libz.tbd (found version "1.2.11")
-- Found PNG: /usr/local/lib/libpng.dylib (found version "1.4.12")
-- Found Python3: /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 (found version "3.9.5") found components: Interpreter Development Development.Module Development.Embed
Checking dependencies:
  libfive:		✓
  Guile bindings:	✘   (skipping)
  Python bindings:	✓
  Studio:		✓   (Python only)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/moritz/code/libfive/build
```

```
ninja -v        
```

I truncate the output; it's about 10 pages more of these and 2ndary ones from the missing types.

```
[1/28] /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Istudio/Studio_autogen/include -I../studio/include -I/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -I../libfive/src/../include -isystem /usr/local/include -isystem /Library/Frameworks/Mono.framework/Headers -isystem /usr/local/Cellar/eigen/3.3.9/include/eigen3 -iframework /usr/local/Cellar/qt@5/5.15.2/lib -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtWidgets.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtCore.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/./mkspecs/macx-clang -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtOpenGL.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtConcurrent.framework/Headers -Wall -Wextra -g -fPIC -pedantic -Werror=switch -march=native -O3 -DNDEBUG -O3 -DEIGEN_NO_DEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -fPIC -std=gnu++17 -MD -MT studio/CMakeFiles/Studio.dir/src/axes.cpp.o -MF studio/CMakeFiles/Studio.dir/src/axes.cpp.o.d -o studio/CMakeFiles/Studio.dir/src/axes.cpp.o -c ../studio/src/axes.cpp
FAILED: studio/CMakeFiles/Studio.dir/src/axes.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Istudio/Studio_autogen/include -I../studio/include -I/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -I../libfive/src/../include -isystem /usr/local/include -isystem /Library/Frameworks/Mono.framework/Headers -isystem /usr/local/Cellar/eigen/3.3.9/include/eigen3 -iframework /usr/local/Cellar/qt@5/5.15.2/lib -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtWidgets.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtCore.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/./mkspecs/macx-clang -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtOpenGL.framework/Headers -isystem /usr/local/Cellar/qt@5/5.15.2/lib/QtConcurrent.framework/Headers -Wall -Wextra -g -fPIC -pedantic -Werror=switch -march=native -O3 -DNDEBUG -O3 -DEIGEN_NO_DEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.15 -fPIC -std=gnu++17 -MD -MT studio/CMakeFiles/Studio.dir/src/axes.cpp.o -MF studio/CMakeFiles/Studio.dir/src/axes.cpp.o.d -o studio/CMakeFiles/Studio.dir/src/axes.cpp.o -c ../studio/src/axes.cpp
In file included from ../studio/src/axes.cpp:20:
In file included from ../studio/include/studio/color.hpp:21:
In file included from /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/QColor:1:
/usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/qcolor.h:342:15: error: 'maybe_unused' attribute cannot be applied to types
    constexpr Q_DECL_UNUSED QColor Color0      {QColor::Rgb, 0xff * 0x101, 0x00 * 0x101, 0x00 * 0x101, 0x00 * 0x101};
              ^
/usr/local/include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
                          ^
In file included from ../studio/src/axes.cpp:20:
In file included from ../studio/include/studio/color.hpp:21:
In file included from /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/QColor:1:
/usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/qcolor.h:343:15: error: 'maybe_unused' attribute cannot be applied to types
    constexpr Q_DECL_UNUSED QColor Color1      {QColor::Rgb, 0xff * 0x101, 0xff * 0x101, 0xff * 0x101, 0xff * 0x101};
              ^
/usr/local/include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
                          ^
In file included from ../studio/src/axes.cpp:20:
In file included from ../studio/include/studio/color.hpp:21:
[... 129 lines cut ...]
              ^
/usr/local/include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
                          ^
In file included from ../studio/src/axes.cpp:20:
In file included from ../studio/include/studio/color.hpp:21:
In file included from /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/QColor:1:
/usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/qcolor.h:359:15: error: 'maybe_unused' attribute cannot be applied to types
    constexpr Q_DECL_UNUSED QColor DarkMagenta {QColor::Rgb, 0xff * 0x101, 0x80 * 0x101, 0x00 * 0x101, 0x80 * 0x101};
              ^
/usr/local/include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
                          ^
In file included from ../studio/src/axes.cpp:20:
In file included from ../studio/include/studio/color.hpp:21:
In file included from /usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/QColor:1:
/usr/local/Cellar/qt@5/5.15.2/lib/QtGui.framework/Headers/qcolor.h:360:15: error: 'maybe_unused' attribute cannot be applied to types
    constexpr Q_DECL_UNUSED QColor DarkYellow  {QColor::Rgb, 0xff * 0x101, 0x80 * 0x101, 0x80 * 0x101, 0x00 * 0x101};
              ^
/usr/local/include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
                          ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
```

mkeeter, 2021-05-15:
No idea!  All of those errors are in Qt headers, so I'd recommend trying to build a simple Qt sample app (e.g. something from [here](https://doc.qt.io/qt-5/qtexamplesandtutorials.html)) that uses the same headers, to isolate the problem.

ether42, 2021-08-22:
I had a similar issue. I'm building like this:
```
cmake -GNinja -DCMAKE_PREFIX_PATH="$(brew --prefix)"/opt/qt\@5/lib/cmake ..
```

The maybe unused attribute from the OP:
```
In file included from .../opt/qt@5/lib/QtGui.framework/Headers/QColor:1:
.../opt/qt@5/lib/QtGui.framework/Headers/qcolor.h:359:15: error: 'maybe_unused' attribute cannot be applied to types
    constexpr Q_DECL_UNUSED QColor DarkMagenta {QColor::Rgb, 0xff * 0x101, 0x80 * 0x101, 0x00 * 0x101, 0x80 * 0x101};
              ^
.../include/QtCore/qcompilerdetection.h:1142:27: note: expanded from macro 'Q_DECL_UNUSED'
#  define Q_DECL_UNUSED [[maybe_unused]]
```

Or even:
```
.../opt/qt@5/lib/QtCore.framework/Headers/qsettings.h:182:5: error: division by zero in preprocessor expression
#if QT_CONFIG(textcodec)
    ^~~~~~~~~~~~~~~~~~~~
.../include/QtCore/qglobal.h:77:30: note: expanded from macro 'QT_CONFIG'
#define QT_CONFIG(feature) (1/QT_FEATURE_##feature == 1)
                            ~^~~~~~~~~~~~~~~~~~~~~
```

Looking around, people encountered this [before](https://gist.github.com/marksteve/cdd263862d5fcdcdb85c383e3d87cd2e#gistcomment-3704827) and you only need to:
```
brew uninstall qt
```

From my limited understanding, there is a conflict between `qt` (currently Qt 6) and `qt@5` (Qt 5, for libfive) and uninstalling  Qt 6 solves it.

I guess https://github.com/libfive/libfive/blob/897b2b35aad958e31ebaec27ae2f805993b37002/README.md?plain=1#L96 should be changed to `brew install qt@5`.