libfive/libfive #446: Loading custom mesh data? (question)

whitwhoa, 2021-05-21:
I've been digging around the source for a bit and can't seem to find a method for generating a custom object from external mesh data. I see a `saveSTL`, but no `loadSTL`. I then dug around `tree.hpp` and `tree.cpp` as these look to be the core objects that are storing what would be vertices and indices, but did not find what I was looking for (although I very well could have missed something). 

How does one go about loading custom mesh data (from a file / array of vertices / etc)?

mkeeter, 2021-05-21:
The internal representation for shapes in `libfive` is a [functional representation](https://en.wikipedia.org/wiki/Function_representation), rather than a mesh, so there's not an easy answer.

It would be possible to import meshes through the [oracle](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/oracle/oracle.hpp#L23-L26) interface, which is designed to interface with the outside world.  That's a bit of a research project – you'd need to write efficient mesh distance query functions for points and intervals – and thus far, no one has taken it on in the open-source community.

whitwhoa, 2021-05-25:
That makes sense. Thank you for the reply. I've just begun researching constructive solid geometry, so implementing "efficient mesh distance query functions for points and intervals" is probably a tad bit outside of my wheelhouse at the moment. Very intrigued by this library and will check back in the future. :)

closed 2021-05-25 by whitwhoa