libfive/libfive #447: Build failure Fedora Core 34

makeclean, 2021-05-22:
### Description

Compile Failure
FC34 - GCC 11.1 - Guile 3.07

I had to build guile from scratch since FC34 comes with guile < 2.2 - seemed to build fine

```
[ 69%] Generating libfive/kernel.go
Backtrace:
In ice-9/psyntax.scm:
 279: 19 [scan ((# #) #(syntax-object *unspecified* # #)) () (()) ...]
In ice-9/boot-9.scm:
3622: 18 [process-use-modules ((#) (#) (#) (#) ...)]
 710: 17 [map #<procedure 7fc68c990800 at ice-9/boot-9.scm:3622:25 (mif-args)> #]
3623: 16 [#<procedure 7fc68c990800 at ice-9/boot-9.scm:3622:25 (mif-args)> (#)]
2900: 15 [resolve-interface (libfive vec) #:select ...]
2825: 14 [#<procedure 7fc68c9830e0 at ice-9/boot-9.scm:2813:4 (name #:optional autoload version #:key ensure)> # ...]
3101: 13 [try-module-autoload (libfive vec) #f]
2412: 12 [save-module-excursion #<procedure 7fc67e44e840 at ice-9/boot-9.scm:3102:17 ()>]
3121: 11 [#<procedure 7fc67e44e840 at ice-9/boot-9.scm:3102:17 ()>]
In unknown file:
   ?: 10 [primitive-load-path "libfive/vec" ...]
In ice-9/eval.scm:
 505: 9 [#<procedure 7fc68cb6f580 at ice-9/eval.scm:499:4 (exp)> #]
In ice-9/psyntax.scm:
1107: 8 [expand-top-sequence ((use-modules # # #)) () ((top)) ...]
 990: 7 [scan ((use-modules (ice-9 textual-ports) (oop goops) ...)) () ...]
 279: 6 [scan ((# #) #(syntax-object *unspecified* # #)) () (()) ...]
In ice-9/boot-9.scm:
3622: 5 [process-use-modules ((#) (#) (#))]
 712: 4 [map #<procedure 7fc68c990800 at ice-9/boot-9.scm:3622:25 (mif-args)> #]
3623: 3 [#<procedure 7fc68c990800 at ice-9/boot-9.scm:3622:25 (mif-args)> (#)]
2903: 2 [resolve-interface (ice-9 textual-ports) #:select ...]
In unknown file:
   ?: 1 [scm-error misc-error #f ...]
In ice-9/boot-9.scm:
 109: 0 [#<procedure 7fc67f1eadc0 at ice-9/boot-9.scm:100:6 (thrown-k . args)> misc-error ...]

ice-9/boot-9.scm:109:20: In procedure #<procedure 7fc67f1eadc0 at ice-9/boot-9.scm:100:6 (thrown-k . args)>:
ice-9/boot-9.scm:109:20: no code for module (ice-9 textual-ports)
make[2]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/build.make:90: libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [CMakeFiles/Makefile2:319: libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2
make: *** [Makefile:149: all] Error 2

```

- Operating system: Fedora Core 34
- `libfive` commit hash: cdc363d0db9e8f14679df4ece2d798c36dc0119c
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): [here]

mkeeter, 2021-05-23:
Unfortunately, I don't have the resources to support Linux distros other than "the latest Ubuntu LTS".  If you can figure this out, I'd be happy to accept a patch, but otherwise, consider building without Guile support by adding `-DBUILD_GUILE_BINDINGS=OFF` to the `cmake` call.

makeclean, 2021-05-23:
Thanks for the quick reply, I did that originally, but it wasn't obvious to me how to switch a guile based example to a python one