libfive/libfive #448: Enable the github "discussion" feature?

traverseda, 2021-05-23:
Libfive is fairly new, and I have a ton of questions but I don't want to open issues for each one. Github has a [discussions](https://docs.github.com/en/discussions) feature that might be worth enabling, for things that aren't really worth opening an issue for.

Also the github wiki for this project seems to be broken, which makes sourcing community documentation a bit more difficult.

mkeeter, 2021-05-23:
Good idea, done!

closed 2021-05-23 by mkeeter

traverseda, 2021-05-23:
There also seems to be an issue with the wiki, clicking the wiki redirects me back to the main "code" tab. I imagine that has something to do with wiki permissions?

mkeeter, 2021-05-23:
Yes, it looks like the wiki is configured so that only folks with push permission can edit it.  I'd be worried about opening it up (due to spam), but maybe I can periodically push canonical discussions to wiki pages.

traverseda, 2021-05-23:
I can't even view it. Try looking at it in private browsing?

mkeeter, 2021-05-23:
I just made an empty page for it.  Oddly, if I click "Wiki" from the repo, it doesn't load, but if I open the [link](https://github.com/libfive/libfive/wiki) in a new tab, it _does_ – maybe a Github issue?

traverseda, 2021-05-23:
Seems to work both ways for me now. Weird that that not having a home page, and not having edit rights, would result in that behavior.

I don't think spam is likely to be a problem, other projects with publicly editable wikis don't seem to have any problem with it, but being cautious about it probably won't hurt.

Thanks for enabling this! It's nice to be able to make issues as I start to figure stuff out, even if that's just documenting my own progress and thoughts about something. The issue I opened about points of interest (#444) would probably have been better as a discussion than a feature request, as it the moment it's under-developed. I need to look into how gltf/glb does it so I can make a stronger, better documented, case for how it would work.

So the discussions will be great for half-baked ideas.