libfive/libfive #458: Domain repetition support

Limeth, 2021-07-15:
Hello, I think domain repetition might be a desirable feature, because it would allow for efficient duplication of shapes.
Assuming the evaluation of the `min`imum of N **standard** repetitions requires the evaluation of all N elements (the same shape but translated) of that minimum, then the evaluation of N **domain** repetitions requires the evaluation of that shape just once.

I tried the following, but it doesn't seem like the `floor-remainder` function is supported by shapes.
```scheme
(define (repeat-x shape max-x)
    (remap-shape (shape x y z)
      (floor-remainder x max-x)
      y
      z
  ))
```
This function would repeat the shape within the interval [0; `max-x`) on the x axis, infinitely.
If we wanted a finite number of repetitions, something like `(floor-remainder (max 0 (min (* max-x N) x)) max-x)` instead

In case C0 continuity is required for f-rep shapes (is it?), then maybe there could be an alternative version of `floor-remainder` which would mirror every other repetition, ensuring continuity at the boundary between repetitions.

omgitsraven, 2021-07-15:
Try `modulo` in place of `floor-remainder`. (I'm not sure if it'll correctly handle negative values, but it's a start...)

Limeth, 2021-07-15:
In regular Guile Scheme, the `modulo` function seems to fail on non-integer arguments, otherwise it seems to work like `floor-remainder`. In the scope of a shape definition, however, it seems like `modulo` works fine with a non-integer modulus, and precisely how I would expect `floor-remainder` to work.

As a side note, I tried domain-repeating half of a sphere and I did run into C0 continuity issues, which is expected:
```
(define (repeat-x shape max-x)
    (remap-shape (shape x y z)
      (modulo x max-x)
      y
      z
  ))

(repeat-x (move (sphere 1) [1 0 0]) 1.0)
```
[image]