libfive/libfive #462: Build failure with Eigen 3.4

ether42, 2021-08-22:
### Description

Apparently, Eigen 3.4 was released a [few days ago](https://eigen.tuxfamily.org/index.php?title=3.4) and there is an incompatible change when using C++ 17. I'm opening this issue to let you know (example patch below).

I'm building like this:
```shell
cmake -GNinja -DCMAKE_PREFIX_PATH="$(brew --prefix)"/opt/qt\@5/lib/cmake ..
ninja
```

And it fails for multiple classes with:
```
In file included from .../libfive/libfive/src/render/brep/simplex/simplex_object_pool3.cpp:11:
.../libfive/libfive/src/render/brep/simplex/../object_pool.inl:161:20: error: no member named 'operator delete[]' in 'libfive::SimplexTree<3>'
                T::operator delete [](fresh_blocks[j].first);
                   ^
.../libfive/libfive/src/render/brep/simplex/../object_pool.inl:48:40: error: no member named 'operator new[]' in 'libfive::SimplexTree<3>'
                    static_cast<T*>(T::operator new[](sizeof(T) * N)), 0));
```

This is documented at https://eigen.tuxfamily.org/index.php?title=3.4#Improvements_to_Eigen_Core:
>  Eigen now uses the c++11 alignas keyword for static alignment. Users targeting C++17 only and recent compilers (e.g., GCC>=7, clang>=5, MSVC>=19.12) will thus be able to completely forget about all issues related to static alignment, including EIGEN_MAKE_ALIGNED_OPERATOR_NEW. 

Especially, you can see `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` is now a noop: https://gitlab.com/libeigen/eigen/-/blob/3.4/Eigen/src/Core/util/Memory.h#L785

### Versions

- Operating system: macOS Catalina 10.15.7 + Homebrew for libraries (`brew install eigen` installs version 3.4)
- `libfive` commit hash: 601730dc
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.15.2 (Homebrew)

### Additional Information

This appears to be the simplest patch to fix the issue: https://github.com/ether42/libfive/commit/d009c90cc9a5ee2af11620b58c653e4eb7772259

I didn't make a pull request because I've no idea whether alignment could still be an issue for people using older versions of Eigen. Feel free to cherry-pick if not.

### Common questions

If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?
Yes (with above patch, minus flaky test from #424).

E1k3, 2021-09-17:
Did you have luck running the binaries resulting from your patch? I seem to get segfaults whenever I run Studio or the libfive-test binary.

mkeeter, 2021-09-17:
I'd like to keep things building on the latest Ubuntu LTS (which has Eigen 3.3), so I think we'll need conditional definitions to enable / disable the `EIGEN_MAKE_ALIGNED_OPERATOR_NEW`.  I'll look into this when I have time, but not sure when that will be!

doug-moen, 2021-10-20:
Also reported in bug #470 and fixed in PR #473