libfive/libfive #464: Segfault on libfive-test and Studio

E1k3, 2021-09-17:
### Description

Running either the tests or the Studio binary results in a segmentation fault.

Studio also prints a guile error. Sometimes in addition to the segfault, sometimes just the guile error on its own.
Studio:
```
guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
```
or
```
guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
[1]    356695 segmentation fault (core dumped)  ./Studio
```

libfive-test:
```
$ ./libfive/test/libfive-test

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[1]    353979 segmentation fault (core dumped)  ./libfive/test/libfive-test
```

gdb libfive-test:
```
(gdb) run
Starting program: ~/libfive-test
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7ffff733d640 (LWP 354099)]
[New Thread 0x7ffff6b3c640 (LWP 354100)]
[New Thread 0x7ffff633b640 (LWP 354101)]
[New Thread 0x7ffff5b3a640 (LWP 354102)]

Thread 2 "libfive-test" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff733d640 (LWP 354099)]
0x00007ffff7e4ac84 in _mm256_store_pd (__A=..., __P=0x7ffff7fa60d0) at /usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/include/avxintrin.h:875
875	  *(__m256d *)__P = __A;
```

### Versions

- Operating system: ArchLinux
- `libfive` commit hash: Kevin Boulain's patch for #462 (https://github.com/ether42/libfive/commit/d009c90cc9a5ee2af11620b58c653e4eb7772259) on top of 601730dc
  - Is this the latest commit to `master`?  yes (plus the patch)
- Qt version: qt5-base 5.15.2
- Eigen version: eigen 3.4.0
- LibPNG version: libpng 1.6.37
- Boost version: boost 1.76.0
- Guile version: guile 2.2.7
- Python version: python 3.9.7

tachiniererin, 2021-10-05:
I'm getting the same results, though i added ifdefs to the patches by ether42 (see here https://github.com/tachiniererin/libfive/tree/fix-eigen34-compat) so the line numbers might not mach.

```gdb
#0  0x00007ffff46c8d22 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff46b2862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff46b2747 in __assert_fail_base.cold () from /usr/lib/libc.so.6
#3  0x00007ffff46c1616 in __assert_fail () from /usr/lib/libc.so.6
#4  0x00007ffff7159434 in Eigen::internal::plain_array<double, 12, 0, 32>::plain_array (this=0x7fff4c015b50) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128
#5  0x00007ffff7156a6c in Eigen::DenseStorage<double, 12, 3, 4, 0>::DenseStorage (this=0x7fff4c015b50) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:211
#6  0x00007ffff7152904 in Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 4, 0, 3, 4> >::PlainObjectBase (this=0x7fff4c015b50) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:476
#7  0x00007ffff714d8da in Eigen::Matrix<double, 3, 4, 0, 3, 4>::Matrix (this=0x7fff4c015b50) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:259
#8  0x00007ffff714a5b0 in libfive::DCLeaf<3u>::DCLeaf (this=0x7fff4c015b30) at /home/elen/code/libfive/libfive/src/render/brep/dc/dc_tree.inl:66
#9  0x00007ffff718a4c2 in libfive::ObjectPool<libfive::DCLeaf<3u>, libfive::Intersection<3u> >::get<>() (this=0x7fff5fffe1c0) at /home/elen/code/libfive/libfive/src/render/brep/dc/../object_pool.inl:60
#10 0x00007ffff7146a57 in libfive::DCTree<3u>::evalLeaf (this=0x7fff4c000fc0, eval=0x7fff601880c0, tape=std::shared_ptr<libfive::Tape> (use count 118, weak count 1) = {...}, object_pool=..., neighbors=...)
    at /home/elen/code/libfive/libfive/src/render/brep/dc/dc_tree.inl:264
#11 0x00007ffff70b6e93 in libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::run (eval=0x7fff601880c0, tasks=..., root=..., root_lock=..., settings=..., done=...)
    at /home/elen/code/libfive/libfive/src/render/brep/dc/../worker_pool.inl:195
#12 0x00007ffff70b724f in libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}::operator()() const (
    __closure=0x7fffdc034f38) at /home/elen/code/libfive/libfive/src/render/brep/dc/../worker_pool.inl:70
#13 0x00007ffff70c0690 in std::__invoke_impl<void, libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}>(std::__invoke_other, libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}&&) (__f=...) at /usr/include/c++/11.1.0/bits/invoke.h:61
#14 0x00007ffff70c0645 in std::__invoke<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}>(libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}&&) (__fn=...) at /usr/include/c++/11.1.0/bits/invoke.h:96
#15 0x00007ffff70c05f2 in std::thread::_Invoker<std::tuple<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0x7fffdc034f38) at /usr/include/c++/11.1.0/bits/std_thread.h:253
#16 0x00007ffff70c05c6 in std::thread::_Invoker<std::tuple<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >::operator()() (this=0x7fffdc034f38) at /usr/include/c++/11.1.0/bits/std_thread.h:260
#17 0x00007ffff70c04a6 in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >, void>::operator()() const (this=0x7fff5fffec30) at /usr/include/c++/11.1.0/future:1385
#18 0x00007ffff70c031b in std::__invoke_impl<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >, void>&>(std::__invoke_other, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<libfive::WorkerPool<libfive::DCTree<3u>, libfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >, void>&) (__f=...) at /usr/include/c++/11.1.0/bits/invoke.h:61
#19 0x00007ffff70bff35 in std::__invoke_r<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>, std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_ [... cut ...] bfive::DCNeighbors<3u>, 3u>::build(libfive::Evaluator*, libfive::Region<3u> const&, libfive::BRepSettings const&)::{lambda()#1}> >, void>*> > >::_M_run() (this=0x7fffdc0030d0) at /usr/include/c++/11.1.0/bits/std_thread.h:211
#38 0x00007ffff4a8a3c4 in std::execute_native_thread_routine (__p=0x7fffdc0030d0) at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:82
#39 0x00007ffff5ec1259 in start_thread () from /usr/lib/libpthread.so.0
#40 0x00007ffff478a5e3 in clone () from /usr/lib/libc.so.6

```

tachiniererin, 2021-10-05:
I tested my fix in #469 with clang 12.0.1 and GCC 11.1.0 and all tests except two with GCC and 3 with clang pass.

mkeeter, 2021-10-07:
Fixed in the commit above (hopefully 😅)