libfive/libfive #472: SEGV (null ptr read) in multithreaded evaluator

doug-moen, 2021-10-11:
The code that Matt provided in discussion #468 causes a crash in the libfive evaluator if multithreading is enabled.

I am running Fedora Workstation linux.

There will be a separate pull request containing a failing test case added to the libfive test suite.

doug-moen, 2021-10-15:
The bug is actually in the documentation. In `libfive/include/libfive/render/brep/settings.hpp`, in `struct BRepSettings`, it says:
```
    /*  Number of worker threads to use while meshing.  Set as 0 to use the
     *  platform-default number of threads. */
    unsigned workers;
```
If you actually set `workers` to 0 then the evaluator crashes. Two suggested fixes:
1. Change the documentation to match the behaviour. `workers` must be >= 0.
2. Change the behaviour to match the documentation. If `workers==0` then set workers to `std::thread::hardware_concurrency()` from header `<thread>`.

mkeeter, 2022-02-20:
Thanks, I decided to fix the docs instead of plumbing changes through all of the rendering pipelines.

commit 0640f0ed (2022-02-20): Change docs for worker threads (closes #472)
closed 2022-02-20 by mkeeter