libfive/libfive #477: Emacs?

Kreyren, 2021-12-19:
Is there a way to get the render preview in emacs?

occivink, 2021-12-20:
To use an external editor, use `File > Open as viewer`, then the file will be reloaded everytime it is modified on disk. It would make sense to add a commandline option to directly open in this mode.

Kreyren, 2021-12-20:
> To use an external editor, use `File > Open as viewer`, then the file will be reloaded everytime it is modified on disk. It would make sense to add a commandline option to directly open in this mode. -- @occivink (https://github.com/libfive/libfive/issues/477#issuecomment-997786033)

That doesn't work for me as changing the file in the external editor causes fatal failure:

[image]

FWIW i was told that the inspekt3d is fully integrated in geiser-guile to provide the interactive preview in emacs, but none seems to know how to configure it (asked on emacs's mailing list as well).

Kreyren, 2022-01-11:
CC @bsima -- https://github.com/libfive/libfive/issues/274#issuecomment-735953692

Can you elaborate how are you using inspekt3d in emacs please? I am unable to figured it out and the repo is 404 on gitlab.

Kreyren, 2022-11-03:
Moved in https://github.com/libfive/libfive/discussions/510

closed 2022-11-03 by Kreyren