libfive/libfive #478: Chat? (IRC/Matrix)

Kreyren, 2021-12-20:
It feels like that there is a steep learning curve for libfive.

Care to provide a community chat to make this learning process less painful?

doug-moen, 2021-12-20:
My comments based on experience with the Curv community chat.

We tried IRC. It's fine for people who like IRC, but it has a steep learning curve, and I didn't like it. In order to catch up on conversation I missed while logged out, I was instructed to either "install a server" (each user has to do this), or pay $5/month to a commercial web hosted IRC client. I couldn't just paste an image into a message using CTRL-V: I was instructed to "use an image hosting service" like Imgur and then paste the URL, or again use a commercial ($) IRC client that automates image pasting.

Now we are using Matrix, and it's much better than IRC (IMO). Very easy to use. There is persistent history and simple image pasting. The chat room is hosted on matrix.org for free, I either use a web client (app.element.io) or I use the mobile Element client, and they work well.

The good thing about Matrix is that it seems more popular and accessible than the Google Groups email list I run. There were more messages in one day than in the 5 year history of the Curv mailing list. A lot of useful information has been posted, about using Curv, how it works, recipes for creating signed distance fields, and images showing what the SDFs look like. The discussion about new SDFs would also be interesting to the libfive community.

The bad thing about Matrix is that all that useful information is unreasonably hard to access. It's there in the history, but scrolling backward through thousands of messages to possibly find a solution to your problem is not really feasible. It's not like finding an answer to a question on stackexchange.

How would you feel about a Discourse server for discussing SDF based modelling tools? With multiple channels, for libfive, Curv, general SDF discussion. There's enough common ground between Curv, libfive, and the other SDF modelling tools, that I think it would be fruitful to bring the entire SDF scene to a common meeting place. I've not used Discourse extensively, but it seems to combine the best elements of chat with organizing discussions so that they are findable and google searchable later on.

CC: @lf94

lf94, 2021-12-20:
I wouldn't mind administrating the Discourse server but it needs *at least* a $10/month VPS due to its storage requirements (10GB minimum). It's not something I'd like to pay for out of pocket currently, as I already have some other server-based expenses.

mkeeter, 2021-12-20:
How do folks feel about Discord?  It's definitely got downsides (searchability, not self-hosted), but it's free and very popular among open-source projects.  It also has the advantage that many folks are already using it, so joining a new server is a low barrier to entry.

Kreyren, 2021-12-20:
Discourse has program for Open-Source to provide free hosting so if you want to maintain discourse then just fill up this form https://free.discourse.group/

About chat i propose using Matrix and setting up bridges as needed there are providers such as https://t2bot.io that provide free hosting without guaranteed uptime.

lf94, 2021-12-20:
Ah nice. I will look into this for the community if everyone wants. And yeah we're using t2bot but you can host your own if you want. I would suggest just to use the t2bot.

Kreyren, 2021-12-20:
> How do folks feel about Discord? 

Strongly against as it's privacy-invading data miner [https://www.tosdr.org/en/service/discord] and security disaster waiting to happen [https://www.cybersecasia.net/tips/discord-is-a-malware-disaster-waiting-to-happen].

occivink, 2021-12-20:
I like the idea of discourse for long-lived discussions on the topic of SDF modelling.
An instant discussion platform a la IRC/matrix/discord could be useful too, but it's maybe best not to fragment the community too much.

Kreyren, 2021-12-20:
FWIW from my experience forum and chat has two different usecases.. 

i usually use chat when i don't understand something and need it explained in a short amount of time and for brainstorm which is uncomfortable to do over forum as there is delay and it gets less attention from the community in comparison.

and forum for complicated issues and feature/spec proposals.

mkeeter, 2021-12-20:
Discourse sounds good to me, but I don't have time to set it up / maintain / moderate it.  My preferred strategy is something like [Zig's](https://github.com/ziglang/zig/wiki/Community) – if folks want to run a community SDF group, I'd be happy to link to it, but am unlikely to maintain such a thing myself.

I turned on Github discussions here in the repo as well, for medium-scale discussions.

Kreyren, 2021-12-20:
> Discourse sounds good to me, but I don't have time to set it up / maintain / moderate it. -- @mkeeter (https://github.com/libfive/libfive/issues/478#issuecomment-998067910)

The provided solution in https://github.com/libfive/libfive/issues/478#issuecomment-998054323 outsources the set up and maintenance on discourse upstream. 

Discourse also has moderating bots and is adapted to make it managable to outsource the maintenance on community.

lf94, 2021-12-20:
I think as long as there is endorsement "it will work". I suggest you shut off Github discussions here as discussions should be driven to one place. I'm going to suggest to Doug to also close the Google group and direct users to the Discourse forum.

There are real problems with decentralized communities when community size is initially small, but also it's easier to split-off than to combine like-communities. If libfive or curv or anything grows to a point where it needs its own forum because of all the unique categories of topics it is easily done. I really see Zig's strategy as probably harmful to itself in the short-medium term.

If everyone is ok with it, I can try to apply for an open-source tier. Just give me 3 thumbs up.

lf94, 2021-12-20:
Community name suggestions?... Just "SDF" is too short, and I don't think "SDF Development Force" is going to be liked too much :laughing: . How about "SDF User Group" ?

lf94, 2021-12-20:
I've submitted the application. I went with SDF User Group and the URL is actually sdfug.discourse.group as "sdfug" will be easier to use as a domain name in the future if it grows too big and we have to self-host.

rishabhnambiar, 2021-12-21:
> I've submitted the application.

Accepted, thanks for giving Discourse a try 🚀

lf94, 2021-12-21:
As @rishabhnambiar has said, it's been accepted and now available at https://sdfug.discourse.group/ !

mkeeter, 2021-12-21:
Okay, I added a link to the README.

I'm still going to leave the Github Discussions section open, because it's convenient for low-level discussions, and I'm more likely to check Github notifications than the Discourse forum.

closed 2021-12-21 by mkeeter