libfive/libfive #480: Studio: 2D files render hollow from bounds to bounds

Kreyren, 2021-12-20:
### Description

[image]

### Steps to reproduce
0. Get libfive with `Studio` editor
1. Render the following file:
```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 100)

(polygon 1 4 [0 0])
```

### Expected
Solid 2D object with Z height of 0.01

### Error messages

```
...
(Studio:18371): dconf-WARNING **: 07:27:03.278: failed to commit changes to dconf: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name ca.desrt.dconf was not provided by any .service files
...
```

### Versions

- Operating system: GNU Guix (f80b5c3d2f04ada0cb2534edb12da5a58eed94ce)
- `libfive` commit hash: 0-4.8ca1b86
  - Is this the latest commit to `master`? ->  I have no idea it's not versioned
- Qt version (if applicable): Dunno how to get that

occivink, 2021-12-20:
The reason is that `polygon` defines a shape that is independent from Z, making it infinite in that direction. To give it a finite Z height, you should use extrude-z, like so: `(extrude-z (polygon 1 4 [0 0]) 0 0.01)`

Kreyren, 2021-12-20:
> The reason is that `polygon` defines a shape that is independent from Z, making it infinite in that direction. To give it a finite Z height, you should use extrude-z, like so: `(extrude-z (polygon 1 4 [0 0]) 0 0.01)`

That seems perfect, proposing it to be the default to avoid the need for `0.01`.

mkeeter, 2021-12-20:
That doesn't make sense as a default, because then you wouldn't be able to (for example) cut a 2D polygon out of a 3D shape, e.g. `(difference (sphere 1) (circle 0.2))`

This is a dupe of https://github.com/libfive/libfive/issues/35 ; the right fix would be to apply a different rendering strategy for shapes which only contain X and Y.

closed 2021-12-20 by mkeeter