libfive/libfive #483: Q: Is there a non-complicated way to make trapezoid?

Kreyren, 2021-12-20:
i am trying to define an alluminium profile to design a brace for the Z-Axis on Creality Ender 3 printer:

[image]

with dimensions:
[image]

and i got this so far:
```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 100)

;;;; To be contributed
(define cube-centered box-mitred-centered)
;; NOTE(Krey): Make `(cube)` to display `(cube [0 0 0] [1 1 1])`

(define (extrusion x)
  "FIXME-DOCS"
    (difference
      (cube-centered [x x 0.01] [0 0 0])
      (cylinder (* x 0.10) 0.5 [0 0 -0.25])
    )
)
  
(extrusion 1)

;;(extrude-z (polygon 1 4 [0 0]) 0 0.01)
```

Is there a way to define the trapezoid as **ONE** 2D object (meaning without using other shapes to create the trapezoid) centered on `[0 0 0]` to make it more maintainable by reducing the complexity?

Kreyren, 2021-12-20:
FWIW preferably if i could do something like a one polygon which points can be defined in coordinates for scaling

occivink, 2021-12-20:
I'm not a pro, but for a convex shape like this I would use the intersection of `half_space` shapes defined by each edge. So for a unit cube (in python):
```
extrude_z(
  intersection(
    half_space([1,0,0], [1,0,0]),
    half_space([0,1,0], [0,1,0]),
    half_space([-1,0,0], [-1,0,0]),
    half_space([0,-1,0], [0,-1,0])
  ),
  -1, 1
);
```

For non-convex shapes... well they can be decomposed into convex shapes.

Kreyren, 2021-12-21:
> I'm not a pro, but for a convex shape like this I would use the intersection of half_space shapes defined by each edge. So for a unit cube (in python): -- @occivink (https://github.com/libfive/libfive/issues/483#issuecomment-997937743)

I converted it in scheme and gave it 1.5h of time, but i can't figure out how to make it to work that way:

```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 10)

(define (test x y z)
  "FIXME-DOCS"
  ;;(extrude-z
    (intersection
      (half-space [0 0 7.65] [0 0 0])
      (half-space [0 7.65 0] [0 0 0])
      ;;(half-space [0 0 0] [0 0 0])
      ;;(half-space [0 0 0] [0 0 0])
    )
  ;;0 1)
  )
  
(test 1 2 3)
```

[image]

Where i tried to define coordinates for the center of the alluminium extrusion/profile to define the polygonal shape.

Couldn't also make it to work to define a complex shape to do the trapezoid.

Kreyren, 2021-12-21:
I reworked the shape to utilize mirroring:

[image]

So it doesn't need the trapezoid and should work with just simple cutting of difference

closed 2021-12-21 by Kreyren