libfive/libfive #487: How to use trig functions in remap?

omgitsraven, 2021-12-27:
I just tried doing a little domain deformation in a `remap` call:

```
import studio
studio.set_bounds([-2, -2, -2], [2, 2, 2])
studio.set_quality(8)
studio.set_resolution(10)

from libfive.stdlib import *
from math import *

box_centered([1.4,2,0.05]).remap(
    Shape.X(),
    Shape.Y(),
    Shape.Z()+sin(Shape.X())
)
```

but the console is telling me
```
Traceback (most recent call last):
  File "C:/OtherPrograms/libfive/libfive/build/libfive\bind\python\libfive\runner.py", line 79, in run
    r = eval(f, gs)
  File "<file>", line 12, in <module>
TypeError: must be real number, not Shape
```

The problem is definitely in `sin(Shape.X())`; I can replace it with `sin(0)` and it'll compile.

I'm surprised that I haven't run into this before (maybe the problem is exclusive to the python implementation?)  Is there any other way to add a wave to space like this, other than maybe something like a for-loop of `attract`s?

mkeeter, 2021-12-27:
I think this is because `sin` is coming from `from math import *`, so it's Python's function (which doesn't know about `libfive` shapes).

The `Shape` class has [trig functions as member functions](https://github.com/libfive/libfive/blob/master/libfive/bind/python/libfive/shape.py#L175-L194), so I think
```
Shape.Z()+Shape.X().sin()
```
would work (I'm on a new computer so can't test it myself).

mkeeter, 2021-12-27:
Other than the source code, no 😆 

I'm open to suggestions for how to documenting tips-and-tricks like this, which fall somewhere between docstrings and a full-on tutorial...