libfive/libfive #489: Issue compiling stdlib on Mac due to the name stdlib.h

bramp, 2022-02-07:
### Description

I'm trying to include libfive in my own project. I have a cmake that looks like:

```
set(BUILD_STUDIO_APP OFF)
set(BUILD_GUILE_BINDINGS OFF)
set(BUILD_PYTHON_BINDINGS OFF)

add_subdirectory(libfive)
```

I'm not sure if that's the right way to include it, but doing so I get compilation errors, specifically:

```
[ 97%] Building CXX object libfive/stdlib/CMakeFiles/libfive-stdlib.dir/stdlib.cpp.o
In file included from /Users/bramp/myproject/libfive/libfive/stdlib/stdlib.cpp:9:
In file included from ./stdlib.h:2:
In file included from /Users/bramp/myproject/libfive/libfive/src/../include/libfive.h:13:
In file included from /Users/bramp/myproject/libfive/libfive/src/../include/libfive/tree/tree.hpp:11:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/map:481:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/__tree:14:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/iterator:419:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/__functional_base:17:
/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/new:319:5: error: no member named 'posix_memalign' in the global namespace
  ::posix_memalign(&__result, __alignment, __size);
  ~~^
/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/v1/new:330:5: error: no type named 'free' in the global namespace
  ::free(__ptr);
  ~~^
```

and then a bunch of other similar errors such as:

```
cmath:339:9: error: target of using declaration conflicts with declaration already in scope using ::abs;
...
error: reference to unresolved using declaration unique_ptr<unsigned char, void(*)(void*)> __stat_hold(nullptr, free);
...
error: reference to unresolved using declaration  __status = (unsigned char*)malloc(__nkw);
```

**Steps to reproduce**

I do not get this error if I build libfive on its own. I only get it when I create a parent CMakeList.txt.

### Versions

- Operating system: Mac 12.2
- `libfive` commit hash: b7f2baa5390082691cbd09d90170b9a72e369e3b
  - Is this the latest commit to `master`?  yes

### Additional Information

Now this does seem an unusual error, and I got a [lead here](https://groups.google.com/g/r-sig-mac/c/QmRXs7X-eb8) that implied it was due to having a file named `stdlib.h`.  On a hunch I renamed `stdlib.h` to `fivestdlib.h` (and fixed references to it), and that fixed all my issues, while still compiling via the project's standard build process.

While I perhaps am not including this project correctly, it does seem problematic to have a header file named stdlib.h. Would you consider renaming to libfive_std.h or similar.

mkeeter, 2022-02-07:
Interesting – I guess that's what I get for shadowing a C library name.

I'm busy with other work at the moment, but would happily merge a PR that renames it to `libfive_std.h`.

bramp, 2022-02-07:
ack, I'll send you a PR.

FYI I noticed the functions in stdlib.h are not scoped to the libfive namespace (which in my case caused a conflict between std::move and libfive's move). Would you accept a separate PR to move this into that namespace?

mkeeter, 2022-02-07:
I think that would be fine, but it would be a slightly larger task, since `stdlib.h` is parsed by a very crude parser and used to generate bindings for other languages.

PR #490 (2022-02-10): Rename stdlib.h to libfive_stdlib.h to avoid header filename clashes
Fixes #489

Everything builds, and tests pass:

```
All tests passed (910001 assertions in 183 test cases)
```
closed 2022-02-10 by mkeeter