libfive/libfive #493: Studio: issues compiling with Qt 5.12.2

Piezoid, 2022-02-21:
### Description
Compilation fails at 3 locations, related to usage of Qt APIs: [build.log](https://github.com/libfive/libfive/files/8105208/build.log)

### Versions

- Operating system: Arch Linux
- `libfive` commit hash: 0640f0ed
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.12.2

### Additional Information

This seems to be related to recent changes on master rather than system updates, because my old branch #492 compiles on my system.

<details>
<summary>Applying this patch on master solves the compilation errors.</summary>

```diff
diff --git a/studio/src/documentation.cpp b/studio/src/documentation.cpp
index d392d138..19b2441e 100644
--- a/studio/src/documentation.cpp
+++ b/studio/src/documentation.cpp
@@ -47,7 +47,7 @@ DocumentationPane::DocumentationPane(Documentation docs)
             fs.insert(f, docs[mod][f]);
             tags.insert(f, QString("i%1").arg(tags.size()));
             mods.insert(f, mod);
-            max_name = std::max(f.length(), (long long)max_name);
+            max_name = std::max(f.length(), (decltype(f.length()))max_name); // decltype(f.length()) = int
         }
     }
 
diff --git a/studio/src/shape.cpp b/studio/src/shape.cpp
index 6321a783..5bcf5155 100644
--- a/studio/src/shape.cpp
+++ b/studio/src/shape.cpp
@@ -249,7 +249,7 @@ void Shape::startRender(RenderSettings s)
 
         timer.start();
         running = true;
-        mesh_future = QtConcurrent::run(&Shape::renderMesh, this, s);
+        mesh_future = QtConcurrent::run(this, &Shape::renderMesh, s);  // Weird, acts like Qt6?
         mesh_watcher.setFuture(mesh_future);
 
         next = {s.settings, s.div - 1, s.alg};
diff --git a/studio/src/window.cpp b/studio/src/window.cpp
index e2d026e0..c3908cd2 100644
--- a/studio/src/window.cpp
+++ b/studio/src/window.cpp
@@ -17,6 +17,7 @@ along with this program; if not, write to the Free Software
 Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301, USA.
 */
 #include <QActionGroup>
+#include <QScreen> // L76 calls a methods on a QScreen*
 #include <QFileDialog>
 #include <QProgressDialog>
 #include <QSplitter>
```

</details>
However, this might break compilation on other systems/versions.

mkeeter, 2022-02-22:
Thanks for the patch!  Can you see if the latest commit (03b592cfc0fa3d9b3cf8117c191e0836d88a89a3) builds for you?

Piezoid, 2022-02-22:
Yes, it builds fine. Thanks !

closed 2022-02-22 by Piezoid