libfive/libfive #494: Fail to build on windows with eigen 3.4

xq114, 2022-03-03:
### Description

On windows MSVC doesn't provide C11 aligned_alloc function as well as C++17 std::aligned_alloc. See https://docs.microsoft.com/en-us/cpp/standard-library/cstdlib?view=msvc-170#remarks-6

https://github.com/libfive/libfive/blob/03b592cfc0fa3d9b3cf8117c191e0836d88a89a3/libfive/include/libfive/render/brep/default_new_delete.hpp#L42-L54

Here libfive uses std::aligned_alloc to provide the previous EIGEN_MAKE_ALIGNED_OPERATOR_NEW macro, which has caused the build on windows to fail:

```
6>C:\source\libfive\libfive\src\..\include\libfive/eval/eval_array.hpp(127,5): error C3861: 'aligned_alloc': identifier not found [C:\source\libfive\build\libfive\src\libfive.vcxproj]
```

### Versions

- Operating system: Windows 10
- `libfive` commit hash: 03b592cfc0fa3d9b3cf8117c191e0836d88a89a3
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): [here]

### Additional Information

Maybe the original implementation in Eigen 3.3.x can be used as instead

zalo, 2022-04-26:
`<cstdlib>` doesn't implement `aligned_alloc` in MSVC because it's not compatible with their normal `free` implementation.   
So MSVC made their own `_aligned_malloc` and `_aligned_free` solution instead.

This is currently untested, but a theoretical solution is to add this to the top:
```c++
#if _MSC_VER
#include <malloc.h>
#include <stdio.h>
#endif
```

And this after `#if EIGEN_MAJOR_VERSION == 4` in `default_new_delete.hpp`:
```c++
#if _MSC_VER
#define ALIGNED_OPERATOR_NEW_AND_DELETE(T) \
    void *operator new[](std::size_t size) {                        \
        return _aligned_malloc(size, std::alignment_of_v<T>);       \
    }                                                               \
                                                                    \
    void operator delete[](void* ptr) {                             \
        _aligned_free(ptr);                                         \
    }
#else
// OLD ALIGNED_OPERATOR_NEW_AND_DELETE HERE
#endif
```

(Untested because I'm running into seemingly unrelated Windows build issues...)

mkeeter, 2022-04-26:
Fixed (hopefully) in #498, thanks @zalo !

closed 2022-04-26 by mkeeter