libfive/libfive #497: Memory access violations C++ (Eigen allocator?)

panic-at-the-disko, 2022-04-09:
I am using the C++ library and am running into memory access violations when using certain functions. It seems to happen primarily with vectors created by functions in the libfive namespace. Stepping through my code line by line, the access violations seem to occur on cleanup/when going out of scope (rather than objects are initialized, constructed or called). 

Sample code below, where the vector from the .walk() method is the troublesome object (if this line is commented out, no errors are thrown):
```
int main(){
	{
		//Vars
		float rad {5.0};
		float centerX {10.0};
		float centerY {20.0};
		float centerZ {0.0};

		//Tree
		libfive::Tree radT(rad);
		TreeVec3 centerT{libfive::Tree(centerX), libfive::Tree(centerY), libfive::Tree(centerZ)};
		libfive::Tree testSphere = sphere(radT, centerT);
		std::vector<const libfive::Tree::Data*> walked{testSphere.walk()};

		//Prints
		std::cout << "Math: " << testSphere << std::endl;
		std::cout << "Address: " << testSphere.get() << std::endl;
	} //Access violation here
	return 0;
}
```
If I create a vector myself of trees, no errors are thrown. For example, the following has no issues (prior troublesome object commented out):
```
int main(){
	{
		//Vars
		float rad {5.0};
		float centerX {10.0};
		float centerY {20.0};
		float centerZ {0.0};

		//Tree
		libfive::Tree radT(rad);
		TreeVec3 centerT{libfive::Tree(centerX), libfive::Tree(centerY), libfive::Tree(centerZ)};
		libfive::Tree testSphere = sphere(radT, centerT);
		//std::vector<const libfive::Tree::Data*> walked{testSphere.walk()};

		//Prints
		std::cout << "Math: " << testSphere << std::endl;
		std::cout << "Address: " << testSphere.get() << std::endl;

		//Create vector of trees
		std::vector<libfive::Tree> testVec{radT, radT, radT};
	} 

	return 0;
} 
```
I'm running Win10, Eigen version 3.3.9. For my project within Visual Studio, I'm linking to the libraries (libfive and stdlib) "as built" e.g. I did not rebuild them again after the initial libfive build. 

Reading through some of the other issues, Eigen's aligned allocator has been mentioned once or twice (for example, #349). Not sure if this or something similar is behind what I'm experiencing, or whether I have syntax issues based on a misunderstanding of the libfive source code?

Any help is appreciated, thanks!

mkeeter, 2022-04-09:
Interesting!

I don't think this is related to the Eigen allocator alignment, since that's only used during evaluation, and you're seeing failures manipulating the raw `Tree` objects themselves.

In general, storing the results of `walk` isn't the _best_ idea, because they're pointers into Tree data and can be invalidated if the Tree changes or is deallocated.  However, I can't see why that would be happening in this case: no destructors should be called when the `std::vector` is destroyed, so even if the pointers are invalidated, it shouldn't matter.

I can't reproduce this on my macOS machine, so I'd encourage you to try shuffling order, etc and see if that changes anything.  It would also be helpful to hear more specific details about the memory access violations that you're seeing.

panic-at-the-disko, 2022-04-09:
Thanks for the quick response! 

I'll keep trying a few things, but until then more details on the memory access violation below. 
For the code pasted above, this is the exception thrown: 
`0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.`

It seems like this is usually a problem derived from mismanaging pointers, but I'm not sure where I could be doing that in my code. I also thought it could be a permissions issue, where the pointer is pointing to a protected area of memory, but running Visual Studio as admin produces the same results. 

If the issue isn't related to Eigen, it seems more likely to be related to pointers - below is another example with throws the same access violation, using Deck instead of Data:
```
int main(){
	{
		//Vars
		float rad {5.0};
		float centerX {10.0};
		float centerY {20.0};
		float centerZ {0.0};

		//Tree
		libfive::Tree radT(rad);
		TreeVec3 centerT{libfive::Tree(centerX), libfive::Tree(centerY), libfive::Tree(centerZ)};
		libfive::Tree testSphere = sphere(radT, centerT);
		//std::vector<const libfive::Tree::Data*> walked{testSphere.walk()};

		//Prints
		std::cout << "Math: " << testSphere << std::endl;
		std::cout << "Address: " << testSphere.get() << std::endl;

		//Deck
		libfive::Deck d{libfive::Deck(testSphere)};
		auto t = d.tape;
	} 

	return 0;
}
```
It may or may not be worth noting that the created Deck object doesn't seem to be correct, as it show a few million `num_clauses` and also the presence of an oracle

mkeeter, 2022-04-09:
Yeah, that's weird, and I can't reproduce it in that case either.  The invalid Deck definitely sounds like memory corruption of some kind.

I tried running with `AddressSanitizer` and `UndefinedBehaviorSanitizer`, and neither flagged any issues on my system.

The next step would be to run in a debugger to catch exactly _where_ it's trying to access invalid pointers on your system, ideally getting a backtrace.  You could also try building in debug mode (by using `ccmake` to reconfigure the build), and seeing if that changes any behavior – at the very least, it should make the backtraces less obfuscated.

panic-at-the-disko, 2022-05-05:
Sorry for the wait - it turned out that the issues above were caused by the debug settings in MSVC. All sorted and thanks again!

However, I've run into something which actually does seem to stem from the Eigen aligned allocator. From a high level, I'm trying to extract the vertices from a Mesh object for storage in one of my own data structures: 
```
libfive::Region<3> region({-10.0, -10.0, -10.0}, {10.0, 10.0, 10.0});
libfive::BRepSettings settings;
settings.min_feature = 1 / meshResolution;
auto libfiveMesh = libfive::Mesh::render(tree, region, settings);
for (auto vert : libfiveMesh->verts)
     Vertices.push_back(Point(vert.x(), vert.y(), vert.z()));
```
This code all runs fine, until it comes time for cleanup and the Mesh object goes out of scope. The error thrown in the crash is `De-allocation of non-allocated memory ` and running with `AddressSanitizer` includes the following:

`#1 in Eigen::internal::aligned_free`

I'm a bit stuck, because I've tried the following with no impact on outcome:

- Call `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` in `Brep.hpp` 
- Add `-march=native` to the compilation flags in MSVC command line settings
- Run using MSVC compiler and clang

If you have any insight I'd appreciate it!

panic-at-the-disko, 2022-05-11:
Turned out that this was an issue with my debug settings in MSVC - nothing related to libfive. Closing this and sorry for any confusion!

closed 2022-05-11 by panic-at-the-disko