libfive/libfive #505: Build failure Ubuntu 20.04

redtrumpet, 2022-09-06:
### Description

When building on Ubuntu 20.04, the `make` command produced the following error message:
```
Scanning dependencies of target libfive-guile
[ 63%] Generating libfive/kernel.go
Backtrace:
In ice-9/boot-9.scm:
   222:17 19 (map1 (((libfive lib)) ((libfive vec))))
  3899:31 18 (_ ((libfive lib)))
  3297:17 17 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8 16 (_ _)
In ice-9/boot-9.scm:
  3223:13 15 (_)
In ice-9/threads.scm:
    390:8 14 (_ _)
In ice-9/boot-9.scm:
  3507:20 13 (_)
   2806:4 12 (save-module-excursion #<procedure 55b4c2c5db40 at ice-…>)
  3527:26 11 (_)
In unknown file:
          10 (primitive-load-path "libfive/lib" #<procedure 55b4c2c6…>)
In ice-9/eval.scm:
   721:20  9 (primitive-eval (use-modules (system foreign) (# #) # #))
In ice-9/psyntax.scm:
  1241:36  8 (expand-top-sequence ((use-modules (system foreign) …)) …)
  1233:19  7 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   285:10  6 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3898:20  5 (process-use-modules _)
   222:29  4 (map1 (((system foreign)) ((system foreign-library)) # …))
   222:17  3 (map1 (((system foreign-library)) ((srfi srfi-1)) ((…))))
  3899:31  2 (_ ((system foreign-library)))
   3300:6  1 (resolve-interface (system foreign-library) #:select _ # …)
  1669:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1669:16: In procedure raise-exception:
no code for module (system foreign-library)
make[2]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/build.make:71: libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [CMakeFiles/Makefile2:300: libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2
```

If I use `cmake -DBUILD_GUILE_BINDINGS=OFF` it build fine (without guile bindings).

### Versions

- Operating system: Ubuntu 20.04
- `libfive` commit hash: d83cc22709ff1f7c478be07ff2419e30e024834e
  - Is this the latest commit to `master`?  [yes]

GammaSQ, 2023-05-19:
Same issue here, with some teeny differences in numbers:
```
[ 69%] Built target libfive-test
[ 69%] Generating libfive/kernel.go
Backtrace:
In ice-9/boot-9.scm:
   222:17 19 (map1 (((libfive lib)) ((libfive vec))))
  3899:31 18 (_ ((libfive lib)))
  3297:17 17 (resolve-interface (libfive lib) #:select _ #:hide _ # _ …)
In ice-9/threads.scm:
    390:8 16 (_ _)
In ice-9/boot-9.scm:
  3223:13 15 (_)
In ice-9/threads.scm:
    390:8 14 (_ _)
In ice-9/boot-9.scm:
  3507:20 13 (_)
   2806:4 12 (save-module-excursion #<procedure 5c5580e44660 at ice-…>)
  3527:26 11 (_)
In unknown file:
          10 (primitive-load-path "libfive/lib" #<procedure 5c5580e4…>)
In ice-9/eval.scm:
   721:20  9 (primitive-eval (use-modules (system foreign) (# #) # #))
In ice-9/psyntax.scm:
  1241:36  8 (expand-top-sequence ((use-modules (system foreign) …)) …)
  1233:19  7 (parse _ (("placeholder" placeholder)) ((top) #(# # …)) …)
   285:10  6 (parse _ (("placeholder" placeholder)) (()) _ c&e (eval) …)
In ice-9/boot-9.scm:
  3898:20  5 (process-use-modules _)
   222:29  4 (map1 (((system foreign)) ((system foreign-library)) # …))
   222:17  3 (map1 (((system foreign-library)) ((srfi srfi-1)) ((…))))
  3899:31  2 (_ ((system foreign-library)))
   3300:6  1 (resolve-interface (system foreign-library) #:select _ # …)
  1669:16  0 (raise-exception _ #:continuable? _)

ice-9/boot-9.scm:1669:16: In procedure raise-exception:
no code for module (system foreign-library)
make[2]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/build.make:71: libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [CMakeFiles/Makefile2:300: libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
I'm on 266fb4323f41fb0c9da883b67dcc0b529b96c28d

mkeeter, 2023-05-19:
This is a Guile version issue, see the comment [here](https://github.com/libfive/libfive/issues/537#issuecomment-1484244799).  Can you make sure you're trying to link against Guile 3.0.6 or later?