libfive/libfive #517: `offset` not an attribute of `Shape` in python?

omgitsraven, 2022-12-02:
### Description

if I write
```
sphere(1).offset(0.1)
```
it gives an error
```
AttributeError: 'Shape' object has no attribute 'offset'
```

This is easy enough to work around by just changing it to `offset(sphere(1),0.1)` of course, but is there any reason that this can't be chained together in this way like the rest of the stdlib functions can?  (I'm assuming it's just an oversight...)

I looked in the source myself but I think I've lost track of how to edit the stdlib these days, sorry :)

### Versions

- Operating system: Win10
- `libfive` commit hash: 72f2a93
  - Is this the latest commit to `master`?  yes

mkeeter, 2022-12-02:
It's been a while since I've edited the stdlib myself, but I think you could probably add it to the hot-patching list [here](https://github.com/libfive/libfive/blob/master/libfive/bind/python/libfive/shape.py#L290-L292)!

omgitsraven, 2022-12-02:
That did the trick, thank you :)

closed 2022-12-02 by omgitsraven