libfive/libfive #519: Could libfive's core concepts be implemented by a non-expert?

lf94, 2023-01-25:
Hey Matt! Big love for libfive. I think it achieves some great things in the space.

My biggest question is do you think libfive could be re-implemented by a non-expert? A huge ideological thorn in my head is that if anyone is not able to re-implement a concept, no one should marry themselves to it. Self-sufficiency is big for me.

Could you please list the core concepts that someone would need to learn to implement a minimal version of libfive? Are all these concepts well explained?

Thank you very much for your great work!!

doug-moen, 2023-01-25:
Are you looking for information on how to build a system that is like libfive Studio, that allows you to build and render 3D models described as signed distance functions? There are many projects like this, it's not that hard, and I've done it myself.

Or are you looking for information on how to recreate the specific and unique algorithm that Libfive uses to render a signed distance function as a triangle mesh? This AFAIK has only been implemented by Matt. Other projects use different algorithms that are well documented and have been successfully implemented by many people.

lf94, 2023-01-25:
Hey Doug, hope all is well! I guess you've forgotten my username :sweat_smile:  Yeah yeah, building and rendering 3D models as SDFs we know is well within non-experts grasp. Yeah, the biggest is turning SDFs into triangle meshes. From what I understand libfive's algorithm is pretty much the best, but it seems difficult to understand. I was wondering if Matt could offer some advice here :slightly_smiling_face: 

(P.S. I'm hoping to see you pick up curv again!!)

lf94, 2023-01-25:
@mkeeter Sorry, I should have started this as a discussion; bad habit.