libfive/libfive #521: One change to compile with GCC-12

lf94, 2023-01-30:
While updating all the dependencies of curv, I got stuck on a libfive error. This corrects it:

```
diff --git a/libfive/include/libfive/tree/tree.hpp b/libfive/include/libfive/tree/tree.hpp
index a203780f..b7367f3e 100644
--- a/libfive/include/libfive/tree/tree.hpp
+++ b/libfive/include/libfive/tree/tree.hpp
@@ -11,6 +11,7 @@ You can obtain one at http://mozilla.org/MPL/2.0/.
 #include <map>
 #include <memory>
 #include <functional>
+#include <utility>
 #include <unordered_map>
 #include <variant>
 #include <vector>
 ```
 
Because curv vendors its dependencies, I think I'm going to take the time to use this as an opportunity to see how I can get curv to manage its own patchset. In the future it's very possible that other dependencies stop updating or slow on updates, so having a patchset to manage will be easier to keep curv alive in the long run.

mkeeter, 2023-01-30:
Is this not already in the code?
https://github.com/libfive/libfive/blob/master/libfive/include/libfive/tree/tree.hpp#L15

(added in dfc0c74e460fcabf68e0928f7629e3b6701e0136)

lf94, 2023-01-30:
Interesting, I guess the submodule update didn't work properly. Thanks for pointing it out 🙂

closed 2023-01-30 by lf94