libfive/libfive #522: Compiling on Windows?

lf94, 2023-01-31:
I'm getting this error, presumably because `aligned_alloc` is available in glibc, but on Windows this is provided by `_aligned_alloc` by `msvcrt`. This was particularly helpful: https://stackoverflow.com/a/67350125

Is there something I can do about this? Simply PR or did I overlook something?

lf94, 2023-01-31:
I see https://github.com/libfive/libfive/pull/498 but find it weird that we're getting the same issues again. I can confirm that this time we're actually using libfive/libfive and not curv3d/libfive, so it's odd.

lf94, 2023-01-31:
FWIW this is an issue when building under MSYS2 with GCC-12

doug-moen, 2023-01-31:
The necessary Windows fix was already created in https://github.com/libfive/libfive/pull/498/commits/7ab4b954f4e93adaecb3bc21f6eac10156e720cc

The problem is that this fix is conditional on `#if _MSC_VER`, which means it is testing for a specific compiler, rather than testing that we are compiling for the Windows environment.

We've seen this problem before with other dependencies, and the solution is to test `defined(_WIN32)` instead.

closed 2023-02-05 by mkeeter

gsohler, 2023-04-16:
Useing MXE i was able to compile for windows without any effort.
But it  built as libfive.dll,whereas i'd like to statically link libfive.
is it possible to configure that by passing a special parameterto libfive in cmake ?

commit 1db3ddc0 (2023-02-05): try to fix #522