libfive/libfive #525: Level of support for hyperbolic trig functions

d01010101, 2023-02-12:
Why the following does not work? It does, though, if `sin` is used instead of `sinh`.

```
remap-shape (shape x y z)
    (/ x (sinh z))
    (/ y (sinh z))
    z)

In function sinh:
wrong-type-arg: Wrong type argument in position 1
```

What is the level of support of `sinh`? https://github.com/libfive/libfive/pull/348 shows "branch has conflicts that must be resolved". Or are there some math difficulties that prevent `sinh` from being used in `remap-shape`?

mkeeter, 2023-02-13:
They aren't supported at all!  You can find the list of operations [here](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/tree/opcode.hpp).

`sinh` is trying to call a Guile built-in function (rather than a `libfive` wrapper), which won't work in the context of `remap-shape`; that's why you're getting a `wrong-type-arg` error.

I don't remember the current state of #348 ; it was blocked for a while waiting for Eigen to release a new version, but it may be possible to rebase that branch now.

d01010101, 2023-02-13:
There are some functions which I find useful for asymptotes.

https://github.com/libfive/libfive/pull/348 says "mkeeter [force-pushed](https://github.com/libfive/libfive/compare/2fd91f1e7bed8e1c2804e5cd46de21e07ee8dda4..fa79aa621ca607d66190ae221dbf20f517a5eee7) the master branch" but the link does not seem to show anything related to hyperbolic functions.

mkeeter, 2023-02-13:
The code is here: https://github.com/libfive/libfive/pull/348/files

You'll probably want to rebase this branch onto the `master` branch to bring it up to date; I can't offer detailed `git` support, but the internet is [full of tutorials](https://www.atlassian.com/git/tutorials/rewriting-history/git-rebase).

d01010101, 2023-02-13:
I have corrected the patch against the current master branch. However, as it turned out, it is not so trivial to configure a second ssh key for the same site so I do not know how to create a pull request or whatever.

I attach a `diff -r -c2` for the moment. If it is not enough tell me.

[libfive-hyperbolic.diff.gz](https://github.com/libfive/libfive/files/10724926/libfive-hyperbolic.diff.gz)

d01010101, 2023-04-04:
Any chance it gets incorporated? It is a second hyperbolic trig patch to libfive slowly becoming obsolete... I am sorry for the format, but given the triviality of the patch, maybe it is enough?