libfive/libfive #530: nodejs bindings

lf94, 2023-03-03:
While not done, I figured it'd be good to open this issue that others can use to track between the two projects.

https://github.com/lf94/node-libfive

The main library bindings are done but not the stdlib bindings. There's also a wrapper I'm working on as I use it.

As a little test I made a shipping box: https://github.com/lf94/node-libfive/blob/main/examples/shipping-box.js

lf94, 2023-03-03:
I took a mental day to chill and do the rest of the mindless work of writing bindings. So the bindings are done. I think it'd be beneficial to get these merged alongside the Python and Guile bindings. Maintaining this in its own project feels like it's going to get lost to time.

lf94, 2023-03-06:
@mkeeter let me know how you'd like to move forward :slightly_smiling_face:

mkeeter, 2023-03-11:
At this point, I'm hesitant to put them into the main repository for fear of them growing stale; it's not something that I would actively be maintaining, and I've had issues with contributions rotting in the past.

Adding a link in the README would be excellent, and if the bindings hit critical mass of users / developers, I'd be happy to reconsider.

lf94, 2023-03-11:
Sounds good to me

closed 2023-03-11 by lf94