libfive/libfive #531: frac function?

lf94, 2023-03-04:
A lot of SDFs rely on a `frac` function to get the fractional part of a number. I've seen this mostly for SDFs that create screws and bolts: https://www.shadertoy.com/view/XtffzX

My own creation requires a bolt as well and I've created such a function but the final piece missing is `frac`! The only alternative I can think of is using one of the many `twirls`.

omgitsraven, 2023-03-04:
You can use `%`, the modulo operator!
Though note that it can produce some pretty glitchy SDFs, due to the completely abrupt discontinuities; you really might prefer using a twirl after all.  (I get use out of `%`, but only in contexts where there's either plenty of empty space bordering the edges, or the edges always happen to line up smoothly; if you're *trying* to create an abrupt edge, modulo can't actually do that right, because it can't actually describe the 'open' faces that those edges leave behind!)

lf94, 2023-03-04:
This is kind of what I figured (for why something like `frac` is not included). I'll give twirl a whirl!

lf94, 2023-03-05:
At first I was like, how with modulus? Then I realized `.mod(1)`. It works damn perfect! I've got bolts again!

[image]

For anyone that needs it:

```js
const mag = (x, y) => sqrt(x.square().add(y.square()));
const thread = (radius_, screw_) => {
  const [x,y,z] = [X(),Y(),Z()];
  const radius = Value(radius_);
  const screw = Value(screw_);
  const rad = mag(x,y).sub(radius);
  const th = (
    rad.sub(
      abs(
       z.mul(screw).sub(atan2(y)(x)).div(Math.PI / 2)
         .mod(1)
         .sub(0.5)
      ).div(screw)
    )
  );
  return th;
};

const bolt = (diameter, height, pitch) =>
  thread(diameter / 2, pitch)
  .intersection(cylinder(diameter + pitch, height));

```

lf94, 2023-03-05:
P.S. The twirl series of functions, while do twirl, do *not* twirl uniformly across the surface. Just a warning to anyone who uses them. They won't look like the twirling shown in my screenshot using my SDF.

closed 2023-03-05 by lf94