libfive/libfive #532: I cant see a Window when opening Libfive Studio

gsohler, 2023-03-06:
Hi,  I just downloaded libfive for curiosity and it compiled without an issue in my FC37 linux Virtual Machine.
After manually setting LD_LIBRARY_PATH i was able to start 'Studio'

However, when starting expect do see a window, but i don't see.
Using ALT-TAB i can see the presence of 'Studio' but i cannot see its window.

in the terminal i see :
QWaylandGLContext: failed to create EGLContext, error=3009

Is there any requirement missing in my place ?

mkeeter, 2023-03-06:
That sounds like an OpenGL issue; what is the OpenGL version available in your VM?

`libfive` targets OpenGL 3.2, so anything earlier won't work.

gsohler, 2023-03-06:
Hi Matt,

Thank you for the prominent help and immediate reply!

https://forums.virtualbox.org/viewtopic.php?t=101304

I just confirmed that Virtualbox cannot do OpenGL 3.1 as of now,
so issue understood.

Maybe it can be a good idea to print a clear warning message to the console instead .

Thank you for your support

closed 2023-03-06 by mkeeter