libfive/libfive #535: Stream meshing immediately to STL?

lf94, 2023-03-11:
Is it possible to stream the mesh immediately to a file? It'd be nice to do this to be able to render massive models without needing lots of memory!

mkeeter, 2023-03-11:
Kinda, but there are a few complexities:
- Meshing builds an **indexed** mesh, e.g. `v[0]` in the example above is a `uint32_t`, rather than a `vec3f`.  This allows us to use less memory during meshing, since vertices are only stored once.  However, an STL is **not** an indexed representation, so we'd have to unpack into individual vertices
- A `PerThreadBRep` exists – like the name implies – on a per-thread basis.  We have many of them being constructed simultaneously; if all of those threads were trying to write to the same file, we've need some kind of locking or arbitration.

In short, this wouldn't be particularly easy, and I'm not totally convinced of the benefit; my intuition is that the majority of memory is allocated during octree construction, and the meshing step doesn't allocate _much_ more.

lf94, 2023-03-11:
> my intuition is that the majority of memory is allocated during octree construction, and the meshing step doesn't allocate much more.

Gotchya. Then there's no real way forward with this without basically writing the octree to a file, manipulating the file, etc... I'll close :)

closed 2023-03-11 by lf94