libfive/libfive #537: Problem to compile  libfive on FC37

gsohler, 2023-03-26:
Hi McKeeter,

sorry for beeing too brief in the previuos post.
I am  aware that guile was needed, but  in installed guile30-dev  package.  did not even imagine, that guile interpreter
is still missing.  after typing guile into the bash,  it got installed.

but now i have a new issue

### Description

[Description of the issue]
Cannot compile libfive on my FC 37
[Steps to reproduce]
git clone libfive
mkdir build
cd build
cmake ..
make

[What did you expect to happen?  What happened?  All the time?]
I expected that libfive would compile but this did not happen
[Are there any error messages printed to the console?]
Yes: last error  which i see is:

[ 69%] Generating libfive/kernel.go
Backtrace:
In ice-9/psyntax.scm:
 279: 19 [scan ((# #) #(syntax-object *unspecified* # #)) () (()) ...]
In ice-9/boot-9.scm:
3622: 18 [process-use-modules ((#) (#) (#) (#) ...)]
 712: 17 [map #<procedure 7fd45d7c5820 at ice-9/boot-9.scm:3622:25 (mif-args)> #]
3623: 16 [#<procedure 7fd45d7c5820 at ice-9/boot-9.scm:3622:25 (mif-args)> (#)]
2900: 15 [resolve-interface (libfive lib) #:select ...]
2825: 14 [#<procedure 7fd45d7b8100 at ice-9/boot-9.scm:2813:4 (name #:optional autoload version #:key ensure)> # ...]
3101: 13 [try-module-autoload (libfive lib) #f]
2412: 12 [save-module-excursion #<procedure 7fd44f29c000 at ice-9/boot-9.scm:3102:17 ()>]
3121: 11 [#<procedure 7fd44f29c000 at ice-9/boot-9.scm:3102:17 ()>]
In unknown file:
   ?: 10 [primitive-load-path "libfive/lib" ...]
In ice-9/eval.scm:
 505: 9 [#<procedure 7fd45d9746e0 at ice-9/eval.scm:499:4 (exp)> #]
In ice-9/psyntax.scm:
1107: 8 [expand-top-sequence ((use-modules # # # ...)) () ((top)) ...]
 990: 7 [scan ((use-modules (system foreign) (system foreign-library) ...)) () ...]
 279: 6 [scan ((# #) #(syntax-object *unspecified* # #)) () (()) ...]
In ice-9/boot-9.scm:
3622: 5 [process-use-modules ((#) (#) (#) (#))]
 710: 4 [map #<procedure 7fd45d7c5820 at ice-9/boot-9.scm:3622:25 (mif-args)> #]
3623: 3 [#<procedure 7fd45d7c5820 at ice-9/boot-9.scm:3622:25 (mif-args)> (#)]
2903: 2 [resolve-interface (system foreign-library) #:select ...]
In unknown file:
   ?: 1 [scm-error misc-error #f ...]
In ice-9/boot-9.scm:
 109: 0 [#<procedure 7fd45d4dedc0 at ice-9/boot-9.scm:100:6 (thrown-k . args)> misc-error ...]

ice-9/boot-9.scm:109:20: In procedure #<procedure 7fd45d4dedc0 at ice-9/boot-9.scm:100:6 (thrown-k . args)>:
ice-9/boot-9.scm:109:20: no code for module (system foreign-library)
make[2]: *** [libfive/bind/guile/CMakeFiles/libfive-guile.dir/build.make:83: libfive/bind/guile/libfive/kernel.go] Error 1
make[1]: *** [CMakeFiles/Makefile2:283: libfive/bind/guile/CMakeFiles/libfive-guile.dir/all] Error 2

=======
i have no clue what this is about. i could not find labels in the text in the local code, so i cannot cross-link the errors to actual code. and debug the code

### Versions

- Operating system: FC37
- `libfive` commit hash: 248c15c57abd2b1b9ea0e05d0a40f579d225f00f
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): Qt5

### Additional Information

Any additional information, configuration or data that might be necessary to reproduce the issue.
no, i dont think so

### Common questions

mkeeter, 2023-03-26:
Thanks for the details!

I think this is the important line:
```
ice-9/boot-9.scm:109:20: no code for module (system foreign-library)
```

The `(system foreign-library)` module was added in [Guile 3.0.6](https://lists.gnu.org/archive/html/guile-devel/2021-04/msg00037.html).

I suspect that on Fedora, you need to install [`guile30`](https://packages.fedoraproject.org/pkgs/guile30/guile30/) to get Guile 3.x, instead of `guile` (which installs the older Guile 2.x series).

gsohler, 2023-03-28:
issue resolved.  problem with FC37 is that GUILE2.0 provides guile and guild programs which are not supported by libfive.
in order  to make libfive  compile need to insteall guile30 (or guile22) instead and make links from guile30 to guile and from guild30 to guild.
better would be to use guild30 or guild22 when available

gsohler, 2023-03-28:
update:
lbfive does not compile with guile 2.2 compiler in my place due  to unspecified language errors.
with guild3.0 and links corectly setup it compiles and starts correctly with the python sphere example

when i load the guile  tutorial,  bit crashes with this error:

------------
[gigl@fedora build]$ ./studio/Studio
QSocketNotifier: Can only be used with threads started with QThread
;;; WARNING: loading compiled file libfive/bind/guile/libfive/kernel.go failed:
;;; In procedure load-thunk-from-memory: incompatible bytecode kind
guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
QMutex: destroying locked mutex
------------
repetitive calls  crash , too because Studio saves the last loaded example

strace ./studio/Studio crashes quite unspectacular

futex(0x7faa2cff8c80, FUTEX_WAKE_PRIVATE, 1) = 1
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\0\0\1\0\f\0\7\0\0\0\1\0\0\0\0\0\f\0\23\0\0\0", iov_len=24}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, MSG_DONTWAIT|MSG_NOSIGNAL) = 24
futex(0x1ffee18, FUTEX_WAKE_PRIVATE, 1) = 1
poll([{fd=3, events=POLLIN}], 1, -1)    = 1 ([{fd=3, revents=POLLIN}])
recvmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\7\0\0\0\0\0$\0\1\0\0\0\16\0\0\0wl_compositor\0\0\0"..., iov_len=2572}, {iov_base="", iov_len=1524}], msg_iovlen=2, msg_controllen=0, msg_flags=MSG_CMSG_CLOEXEC}, MSG_DONTWAIT|MSG_CMSG_CLOEXEC) = 1056
futex(0x1ffee74, FUTEX_WAKE_PRIVATE, 2147483647) = 1
futex(0x1ffee18, FUTEX_WAKE_PRIVATE, 1 <unfinished ...>
+++ exited with 1 +++

installing into /usr/local/bin did help.
is it neeeded to clear cached guile bytecode somewhere ?

mkeeter, 2023-03-29:
(I edited your post to use code fences ``` for code blocks, because the formatting was messed up)

Can you get a backtrace of where the crash occurs?  `QMutex` is not used anywhere in the `libfive` codebase, so I'm puzzled about the error `QMutex: destroying locked mutex`

`In procedure load-thunk-from-memory: incompatible bytecode kind` is a very suspicious error messages; is it possible that it's compiled the bytecode with Guile 2 but is trying to use Guile 3 in the program (or vice versa)?

Bytecode is compiled in a separate CMake pass [here](https://github.com/libfive/libfive/blob/master/libfive/bind/guile/CMakeLists.txt#L34); does running `guile` and  `guild` from the command line use Guile 2 or 3?

gsohler, 2023-03-29:
Hi Matt,
I have seen you edited libfive/bind/guile/libfive/lib.scm for me.
when i start studio, i get:

[gigl@fedora build]$ ./studio/Studio
QSocketNotifier: Can only be used with threads started with QThread
;;; WARNING: loading compiled file libfive/bind/guile/libfive/kernel.go failed:
;;; In procedure load-thunk-from-memory: incompatible bytecode kind
guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
Segmentation fault (core dumped)

--------
And here is the backtrace 
0x00007f57c8355b55 in dri2_create_screen () at /lib64/libEGL_mesa.so.0
#1  0x00007f57c835fdd7 in dri2_initialize_wayland () at /lib64/libEGL_mesa.so.0
#2  0x00007f57c8353868 in dri2_initialize () at /lib64/libEGL_mesa.so.0
#3  0x00007f57c834bdad in eglInitialize () at /lib64/libEGL_mesa.so.0
#4  0x00007f57c84408a7 in QtWaylandClient::QWaylandEglClientBufferIntegration::initialize(QtWaylandClient::QWaylandDisplay*) ()
    at /usr/lib64/qt5/plugins/wayland-graphics-integration-client/libqt-plugin-wayland-egl.so
#5  0x00007f57df29799c in QtWaylandClient::QWaylandIntegration::initializeClientBufferIntegration() () at /lib64/libQt5WaylandClient.so.5
#6  0x00007f57df2a3c28 in QtWaylandClient::QWaylandDisplay::clientBufferIntegration() const () at /lib64/libQt5WaylandClient.so.5
#7  0x00007f57df29ed0a in QtWaylandClient::QWaylandIntegration::createPlatformWindow(QWindow*) const () at /lib64/libQt5WaylandClient.so.5
#8  0x00007f57e2974c90 in QWindowPrivate::create(bool, unsigned long long) () at /lib64/libQt5Gui.so.5
#9  0x00007f57e31e3b25 in QWidgetPrivate::create() () at /lib64/libQt5Widgets.so.5
#10 0x00007f57e31e417f in QWidget::create(unsigned long long, bool, bool) () at /lib64/libQt5Widgets.so.5
#11 0x00007f57e31f135e in QWidgetPrivate::setVisible(bool) () at /lib64/libQt5Widgets.so.5
#12 0x0000000000450721 in Studio::Window::Window(Studio::Arguments) (this=this@entry=0xed8410, args=...) at /home/gigl/git/libfive/studio/src/window.cpp:413
#13 0x0000000000426dff in Studio::App::App(int&, char**) (this=this@entry=0x7ffdb0ed57a0, argc=@0x7ffdb0ed579c: 1, argv=argv@entry=0x7ffdb0ed58e8)
    at /home/gigl/git/libfive/studio/src/app.cpp:40
#14 0x000000000041b507 in main(int, char**) (argc=<optimized out>, argv=0x7ffdb0ed58e8) at /home/gigl/git/libfive/studio/src/main.cpp:50

Hope it helps
================
of course its not nice that i have to link guile3.0 to guile and guild3.0 to guild to make compile work.
guile could actualy catch wrong shared directories with that.
is it possible to use guild3.0 rather than guild when available ?

mkeeter, 2023-04-01:
Your error seems like a general Qt + Wayland error, rather than something that's specific to `libfive`.  Can you try running other Qt + OpenGL apps under Wayland, e.g. something from [this list](https://doc.qt.io/qt-6/examples-widgets-opengl.html)?  What window manager + GPU are you using?

gsohler, 2023-07-02:
i wasnt able to replicate this issue anymore. sorry for that

closed 2023-07-02 by gsohler

athompso, 2024-03-28:
This issue can still be replicated under Ubuntu 20.04.  Only Guile 3.0.1 is available in the repos, not 3.0.6 as mentioned in #505.  
It is buildable under 22.04 LTS, however, which includes Guile 3.0.7.
Perhaps the build instructions need to be updated to indicated that <master> is no longer buildable under 20.04 LTS, but 22.04 LTS instead?  The instructions are otherwise pretty much perfectly fine.