libfive/libfive #538: Embed Libfive in an application - general SDF Question

gsohler, 2023-03-28:
### Description

[Description of the issue] : How to use LibFive in own application:
[Steps to reproduce] : see sample snippet below
[What did you expect to happen?  using Libfive to render from a distance function
[Are there any error messages printed to the console?]
nope, sporadically segfaulted

### Versions

- Operating system: FC37
- `libfive` commit hash: N/A
  - Is this the latest commit to `master`? YES!
- Qt version (if applicable): N/A

=========
I like this SDF thing very much much(its so similar to doug moen;s curv!)

as i  had no success compiling studio yet due to guile issue,  i DID manage to see barebone functionality of libfive

this is my code which i copied  from test directory

    auto cube = max(max(
        max(-(Tree::X() + 1.5),
            Tree::X() - 1.5),
        max(-(Tree::Y() + 1.5),
            Tree::Y() - 1.5)),
        max(-(Tree::Z() + 1.5),
            Tree::Z() - 1.5));

    Tree object(std::make_unique<CubeOracleClause>());

    //  The region is set so we hit where the interesting stuff happens.
    Region<3> r({ -3., -3., -3. }, { 3., 3., 3. });
   
    std::unique_ptr<libfive::Mesh> mesh = Mesh::render(object, r, settings);
    auto m = mesh.get();
    m->saveSTL("cube.stl");

I can see the cube!!!

Question 1: 
How can i use libfive instead to render with an oracle function callback

double distfunc(double, x, double y, double z);

(which returns the altitude above  object surface

Question 2:

My copied code is not 100% stable, i get random segfaults. is it complete or can i do more initialization ?

Thank you for reading!

mkeeter, 2023-03-29:
For using a single function as an oracle, please see [this discussion](https://github.com/libfive/libfive/discussions/468).

I'm assuming the segfaults is #537 and will reply in that thread separately.

gsohler, 2023-04-03:
Hi Matt!
Thank you for the discussion link!
Now I understand how to use the Oracle in C++.
Is there any chance to perform the same in C only with passing a callback function, or is this functionality simply missing in C world ?
Or: is there any chance converting a Tree instance to libfive_tree ?
Thank you

mkeeter, 2023-04-03:
There's no way to implement oracles in plain C, because they use C++ inheritance.

`libfive_tree` is defined with
```cpp
typedef const libfive::Tree::Data* libfive_tree;
```
([here](https://github.com/libfive/libfive/blob/ae8ade5ada97af319f6ea07f06bd8242336171cd/libfive/include/libfive.h#L184)); you can use [`Tree::{acquire, release}`](https://github.com/libfive/libfive/blob/d83cc22709ff1f7c478be07ff2419e30e024834e/libfive/include/libfive/tree/tree.hpp#L195-L202) to convert to and from this inner representation.

closed 2023-04-16 by gsohler