libfive/libfive #540: Feature request: Separate bounds/resolutions for different objects in the same scene?

omgitsraven, 2023-05-10:
How easy would it be for Studio to allow different objects in the scene at the same time to each have their own associated bounds/quality/resolution?  Partly so I can set lower resolution on objects that need it less, but also so I can crop individual objects' bounds to just the space they fill, if I have separate objects fairly far apart...

omgitsraven, 2023-05-13:
I was just picturing something along the lines of:

```
import studio
from libfive.stdlib import *

studio.set_bounds([-2, -2, -2], [2, 2, 2])
studio.set_quality(8)
studio.set_resolution(10)

sphere(1)

studio.set_bounds([1, -2, -2], [5, 2, 2])
studio.set_quality(5)
studio.set_resolution(20)

sphere(1).move([3,0,0])
```

where every time it encounters a new group of `studio.set_*` calls, it treats everything from then on as another self-contained space, and then it just draws them all in the same window.

mkeeter, 2023-05-14:
Good suggestion!

I think the right fix would be to configure bounds and resolution on a **per-shape** basis, then fall back to the top-level `studio` bounds if that's not present.

It's not _fundamentally_ hard, but would require a lot of plumbing to connect from the Python script all the way to the rendering loop.