libfive/libfive #541: Return multiple shapes without using union, in Python?

omgitsraven, 2023-05-10:
In short, I'm asking this one again: https://github.com/libfive/libfive/issues/81 but, for Python :)

The thread actually mentions the use of `yield from`, but I can't figure out how to use it in this context to get that result....

mkeeter, 2023-05-14:
Sure, done in 9b779fec5a199c00657f9386f744503d223ccaad

This changes Python's result handling to accept lists; it iterates over any lists and extracts shapes for individual rendering.

Here's an example:
```python
import studio
studio.set_bounds([-10, -10, -10], [10, 10, 10])
studio.set_quality(8)
studio.set_resolution(10)

from libfive.stdlib import *
[sphere(1), sphere(1, [1,1,1])]
```

omgitsraven, 2023-05-14:
Unfortunately, it now seems to ONLY accept lists, actually?  When I launch Studio, it shows me an empty bounding box until I enclose the `sphere(1)` in square brackets!

mkeeter, 2023-05-14:
> Unfortunately, it now seems to ONLY accept lists, actually? When I launch Studio, it shows me an empty bounding box until I enclose the sphere(1) in square brackets!

good point, fixed in 266fb4323f41fb0c9da883b67dcc0b529b96c28d 😆 

@poulenet I'll think about #540, but I have limited energy to work on big features (or review big PRs, for that matter); after writing a ton of Rust, coming back to a big C++ codebase is somewhat painful!

closed 2023-05-14 by mkeeter

omgitsraven, 2023-05-14:
Perfect now, thank you!