libfive/libfive #543: Double free or corruption error when trying to use libfive

gsohler, 2023-07-02:
### Description

[Description of the issue]
When trying to use internals of livfive i get a double free or corruption error, segfault
[Steps to reproduce]

I use this code to evaluate  the tree only once:

double sdf_func(libfive::Tree &tr, Vector3d pt)
{
	ArrayEvaluator e((Tree(tr))); <- segfaults here
//	double xx =e.value({pt[0], pt[1], pt[2]});
}

Reading the internet i understand, why its not allowed to use a local variable for use in make_shared<Deck> in ArrayEvaluator.cpp
(the memory will be freed once in local scope and once when the std_shared ptr is not used anymore.
However,  in this case i dont know how to fix it, because "Tree" is a variable and not a ptr to variable.

[What did you expect to happen?  What happened?  All the time?]
It should return a value and not segfault.
Are there any other initialisation steps required to make simple ArrayEvaluator Class  not crash ?

[Are there any error messages printed to the console?]
Yes: 
double free or corruption (out)
Aborted (core dumped)

### Versions

- Operating system: FC37
- `libfive` commit hash:  
  - Is this the latest commit to `master`?  no
- Qt version (if applicable): qt5

### Additional Information

Any additional information, configuration or data that might be necessary to reproduce the issue.

lf94, 2023-10-07:
@gsohler It seems we may be hitting the same error :)

gsohler, 2024-02-23:
This error dominantly happens, when the meshing result from Mesh::render is void.
Issue is that the data returned by Mesh::render() already appears freed. When the returned mesh gets out of scope it gets freed
again and double free error is thrown . Could there be some issue with the Mesh desctructor ?

Is there a reason why Mesh::render returns an unique_ptr and not a shared_ptr ?
I have replaced unique_ptr to shared_ptr 12 times and once replaced release() by get()
and it workes much more stable in my place.
Would you accept a PR ?

lf94, 2024-02-23:
Nice work @gsohler :)