libfive/libfive #547: Expose normals in libfive_mesh

lf94, 2023-10-04:
It seems libfive_mesh doesn't include normals that mesh.cpp actually does compute when writing out to a file.

lf94, 2023-10-04:
Which leads me to a nicer high level function here: 

```C
libfive_vec3 libfive_tree_eval_d(libfive_tree t, libfive_vec3 p)
{
    DerivArrayEvaluator e((Tree(t)));
    auto v = e.deriv({p.x, p.y, p.z});
    return {v.x(), v.y(), v.z()};
}
```

But I'm not sure what exactly when to invoke this. Do I need to traverse the whole SDF, wait until I hit a boundary (0), and then .... something?

BrannonKing, 2025-07-14:
FYI, the code to compute the normals that exists in the sphere test is for the center of the triangle, not for the individual vertices in each triangle. Watch out for that. It's the normal for the face, not the vertex.