libfive/libfive #548: Does Mesh::render consume the libfive_tree?

lf94, 2023-10-07:
I'm having an issue right now where I call libfive_tree_render_mesh twice, and on the second call it segfaults. Is this intended? 

Another segfault happens when I call libfive_tree_optimized() and pass the tree to render_mesh.

I'm trying to get to the bottom if my issue is libfive or the FFI library I'm using.

Thanks!

mkeeter, 2023-10-07:
It's not my intention to have `libfive_tree_render_mesh` consume the argument, but it's possible that there's a bug in the code!

lf94, 2023-10-08:
It was mentioned by someone that the problem may actually be something to do with libfive using threads. Maybe this messes with the FFI... Thank you for clarifying the intent! It's probably not libfive then.

closed 2023-10-08 by lf94

gsohler, 2023-10-10:
I have also read in the libfive docs that the c++ code(Mesh::) uses special destructors which free the data once they are not 
used anymore.
Believe once we are able to establish a minimal testcase, mkeeter could easily spot the bug.