libfive/libfive #550: Unioning nothing at the top level causes things to mesh very slow

lf94, 2023-10-15:
Basically I have this: union(nothing, my_complex_shape), which causes the whole thing to mesh very slowly. Any ideas?

mkeeter, 2023-10-15:
How are you defining "nothing"?

lf94, 2023-10-15:
not at my comp. at the moment, but pretty sure it's libfive_tree_const(0)

mkeeter, 2023-10-15:
That would do it – 0 is ambiguously both inside and outside, and `libfive` will take special effort to figure out which one it is!  Instead, pick a big positive value (e.g. 10000), which is unambiguously outside.

lf94, 2023-10-16:
Roger, I'll give that a shot!

Tangent question: have you had issues with FFI and libfive's threading (I think it has threading right)?

mkeeter, 2023-10-16:
Yes, it uses threads for meshing / rendering, but that _should_ be relatively safe – IIRC it spins up a bunch of threads at the beginning of a meshing / rendering operation, then waits for them to join when finishing.  I don't think there are any **persistent** thread pools (but it's been a while since I wrote that code).

lf94, 2023-10-16:
It must be the FFI I'm using then (koffi.dev). When I force libfive to be slow, or evaluate with values I never consume again, everything works. Otherwise, it segfaults. Thanks for letting me know :)

closed 2023-10-19 by lf94