libfive/libfive #552: koffi and ffi-napi both segfault when meshing

lf94, 2023-10-19:
I've now made bindings in both koffi and ffi-napi, which are FFI libraries for nodejs.

Everything works until I start meshing. When I call libfive_tree_render_mesh, it will segfault shortly after. This includes both FFI bindings I've done.

This is making me wonder if there's an inherent problem with libfive's multithreading capabilities.

Is there a way to "turn off" multithreading without having to recompile libfive? (I don't mind, but it'd be nice for those Linux distros that are distributing libfive compiled with this in)

I would love to get to the bottom of this, as this is a huge blocker for SummonScript...

lf94, 2023-10-19:
If not, I propose a minor API update, that allows libfive_tree_render_* to take a settings object like here: https://github.com/libfive/libfive/blob/f3a283b0f88607909eb439d511d502d8811f5770/libfive/test/oracle.cpp#L103-L108

lf94, 2023-10-19:
Re-compiling libfive with all workers set to 1 completely eliminates the segmentation faulting! Please, we need a way to set this from the API :D

mkeeter, 2023-10-22:
If you can get a traceback of the segfault, that would be helpful to debug the crash.

As a workaround, I'd also be happy to accept a PR adding a new `libfive_tree_render_mesh_st` (as a single-threaded variation).

lf94, 2023-10-22:
Excellent, Ill do a PR asap! And I'll try to capture a backtrace.

lf94, 2023-10-22:
https://github.com/lf94/libfive/commit/f1d6e5067a956ee2531971879d95c4c74534800e just testing and I'll open up a PR

lf94, 2023-10-22:
Almost like magic, multi-threading isn't causing segfaults anymore. Looking at the git history I don't think there's anything that would've fixed it from a re-compile. It would still be safe to offer the single-threaded version and then I can expose this on my side for those that do encounter it. I'll close the issue and re-open if it comes back :)

closed 2023-10-22 by lf94