libfive/libfive #562: Feature resolution is unexpectedly impacted by the size of the bounding box

Wulfsta, 2024-02-26:
### Description

Resolution is impacted when changing the size of the bounding box.

#### Steps to Reproduce

Swap the commented `set_bounds` call in the following Studio script:
```
import studio

### SWAP THESE TO OBSERVE IMPACT ###
studio.set_bounds([-10, -10, -10], [10, 10, 10])
#studio.set_bounds([-10, -10, -1], [10, 10, 6])

test_quality = 9
test_resolution = 10

studio.set_quality(test_quality)
studio.set_resolution(test_resolution)

from libfive.stdlib import *

def quality_test():
    norm = Shape.sqrt(Shape.X() * Shape.X() + Shape.Y() * Shape.Y())
    return Shape.nanfill(Shape.sin(1000.0 / (norm)), 1.0)

qt = intersection(
    quality_test(),
    cylinder(9.9, 5)
)

qt
```

Notice that in the first case there are 11 rings generated before artifacts appear, while in the second case there are only three rings. I would not anticipate this behavior as an end user, and it took me a while to realize that it was not the quality or resolution that was causing artifacts in one of my designs.

### Versions

- Operating system: NixOS (Linux)
- `libfive` commit hash: 7af5f43684a8a497ac8610d39f7fca935364a9b9
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 6.6.2

mkeeter, 2024-02-26:
This is a somewhat confusing side-effect of how we pick the max subdivision level in `region.hpp`:

```cpp
    /*
     *  Returns a version of this region that has the level set
     *  based on the given minimum feature.
     *
     *  This lets us do subdivision without worrying that the termination
     *  condition (of a cell side being < min_feature) is dependent on
     *  floating-point accuracy.
     */
    Region<N> withResolution(double min_feature) const {
        const auto min_dimension = (upper - lower).minCoeff();
        const auto level = ceil(fmaxf(0.0f, logf(min_dimension / min_feature)) /
                                logf(2));
        return Region<N>(lower, upper, perp, level);
    }
```

The number of octree subdivisions is set by the lowest dimension, so I think that going from [10, 10] → [-1, 6] must be reducing it in your example.

Maybe it should be set by the largest dimension instead?

Wulfsta, 2024-02-27:
I agree that this should be per unit, and not dependent on the size of the bounding box. It also has a huge impact on exported STL size - settings that produce no artifacts with a smaller bounding box produce significantly larger files (multiple GiBs in my case), whereas a bounding cube produces no artifacts at default settings and order of magnitude smaller files.

junpls, 2025-09-28:
I can confirm this issue from my observations using the [Unity bindings](https://github.com/zalo/libfive-unity) in combination with libfive binaries that I compiled from [e837098](https://github.com/libfive/libfive/commit/e8370983e7bc6d49409affcc34fc70c673cc876f) (latest `master`).

This scene contains nothing but a cube with a side length of 1.5 (shifted up slightly) subtracted from a cube with a side length of 2.
As I tweak nothing but the bounds size, vastly different meshes are generated:

[image]
[image]
[image]
[image]

Even with only the 2x2x2 cube, certain numbers for the bounds can produce overly detailed meshes:

[image]
[image]