libfive/libfive #565: Need precompiled Windows binaries of any kind

athompso, 2024-03-19:
### Description

[Description of the issue]
Missing Win32/Win64 binaries.

[Steps to reproduce]
n/a

[What did you expect to happen?  What happened?  All the time?]
Hard to argue that three platforms are equally supported, when only one is usable by non-developers.

[Are there any error messages printed to the console?]
n/a

### Versions

- Operating system: [here]
Win10 x64 / Win11 x64

- `libfive` commit hash: [here]
n/a
  - Is this the latest commit to `master`?  [yes]
n/a
- Qt version (if applicable): [here]
n/a

### Additional Information

Any additional information, configuration or data that might be necessary to reproduce the issue.

I understand that building and *packaging* Windows software is a royal PITA - I used to do that for a living!  However, shipping bare executables/DLLs in a zip file is not _that_ hard in comparison to a full release, and would at least let the more adventurous users attempt to run Studio.
I'm an ex-developer, and even I don't want to install MSVC on my main system and compile Studio just to view someone's libfive model they want to show me.  (Mind you, I want to go buy a Mac to view it even less, so... hm.)
"Read the source" is somewhat understandable for Linux users, but not really for Windows users.

athompso, 2024-03-19:
Just checked, GitHub still doesn't support automatic MSVC compilation via Actions :-(

athompso, 2024-03-19:
So assuming you can't find any volunteer to build stuff, and you want to spend _some_ time doing the work, this is the best option I know of:  https://developer.microsoft.com/en-us/windows/downloads/virtual-machines/

Now you can feel free to tell me to jump off a cliff :-)

mkeeter, 2024-03-19:
Sorry, but I don't have time to maintain CI for a platform that I'm not using; just getting it building was enough of a struggle!

closed 2024-03-19 by mkeeter

athompso, 2024-03-19:
Understood.  Been there, done that.  Thank your for making it open source in the first place, so others at least enjoy the _possibility_ of running it wherever they want, if they care enough.